z88dk/z88dk #581: z80asm: map file hold incorrect address values for code coming from consolidated object file

aralbrec, 2018-01-30:
I'm looking at a large project that is compiling using a consolidated object file.  The output executable seems to be ok (ie it's correctly linked) but almost all the addresses values listed in the map file are wrong.

If @pauloscustodio can't see something wrong with how the map file is made when a consolidated object file is used during linking, I can maybe zip up the project and send in PM.  It's a large project and takes around 10-12 mins to compile.

aralbrec, 2018-01-30:
An example:

From game.map:
```
game_sprite_c_l_spr_move_left_00102 = $972B ; addr, local, , game, code_compiler, game_sprite.c:1427
game_sprite_c_l_spr_move_left_00104 = $974A ; addr, local, , game, code_compiler, game_sprite.c:1444
game_sprite_c_l_spr_move_left_00119 = $97B8 ; addr, local, , game, code_compiler, game_sprite.c:1512
game_sprite_c_l_spr_move_left_00113 = $97B0 ; addr, local, , game, code_compiler, game_sprite.c:1504
```

For the lis file generated by the compiler:

```
1425  030D  7D          	ld	a,l
1426  030E  32 00 00    	ld	(_tmp0),a
1427  0311              l_spr_move_left_00102:
1428  0311  21 00 00    	ld	hl,_sprite_horizontal_check
1429  0314  36 00       	ld	(hl),0x00
1430  0316  3A 00 00    	ld	a,(_tmp0)
1431  0319  B7          	or	a, a
1432  031A  28 18       	jr	Z,l_spr_move_left_00108
```

At address $9729 in memory:

```
   ld a,l
   ld ($f9d5),a
; address $972d is the actual "l_spr_move_left_00102"
   ld hl,$f9c6
   ld (hl),0
   ld a,($f9d5)
   or a
   jr z,$9750
```

All labels are off by varying amounts.

pauloscustodio, 2018-01-30:
@aralbrec, I did not manage to reproduce the error. Please PM with the project and build instructions so that I can look at the actual error.

aralbrec, 2018-01-30:
PM sent.

aralbrec, 2018-02-11:
The problem was in the batch file that built the game.  Two binaries were produced and two map files with the second map file overwriting the first.  The first binary was being looked at using the second map file so addresses were naturally slightly different.

closed 2018-02-11 by aralbrec