z88dk/z88dk #582: SC3000 reset after running

Fabrizio-Caruso, 2018-01-30:
I have tried to compile CROSS CHASE for the SC-3000.
The compilation is fine.
``` 
	$(Z88DK_PATH)$(MYZ88DK) +sc3000 \
	-O3 \
	-pragma-need=ansiterminal \
	-DTINY_GAME -DNO_SLEEP -DLESS_TEXT -D__GAL__ -DNO_RANDOM_LEVEL -DALT_PRINT -DNO_MESSAGE \
	-clib=ansi \
	-vn -lndos -create-app -Cz--audio \
	-o $(BUILD_PATH)/TINY_sc3000.prg \
	$(SOURCE_PATH)/display_macros.c $(SOURCE_PATH)/enemy.c $(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c $(SOURCE_PATH)/text.c \
	$(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c $(SOURCE_PATH)/main.c
```
If I run mess with the compiled .wav and the basic rom 3.1 version 1:
"Sega_BASIC_Level_3_V1_(SC-3000)"
the game loads (its name is found).
If I then type run, the computer resets to the initial screen.

suborb, 2018-02-01:
My hunch is that this is the same as #585

Fabrizio-Caruso, 2018-02-02:
@suborb, indeed it was it! Thanks!

closed 2018-02-02 by Fabrizio-Caruso