z88dk/z88dk #584: zsdcc: current compile of gandalf project leads to program bugs

aralbrec, 2018-02-01:
The project is here:
https://bitbucket.org/CmGonzalez/gandalf/src

The author is using a version of z88dk from July 2017 that produces a stable result.  The current z88dk produces bugs in the compile.  I think it must be down to new peephole rules introduced since July.

I'd like to fix this one before the release.

aralbrec, 2018-02-10:
The author has updated to the current version and things are compiling fine.  There were bugs in previous versions of the game so maybe one of those was causing crashes

closed 2018-02-10 by aralbrec

pauloscustodio, 2018-02-11:
Does this also closes #581? As far as I could analyse, the map file was not up-to-date with the generated code.

aralbrec, 2018-02-11:
> Does this also closes #581<https://github.com/z88dk/z88dk/issues/581>? As far as I could analyse, the map file was not up-to-date with the generated code

I’ve found the problem just now - the batch file makes two binaries and two map files with the second map file overwriting the first.  So I was looking at the first binary with the second map file and that’s why it was all messed up.