z88dk/z88dk #587: [MICROBEE] Microbee broken in recent versions

Fabrizio-Caruso, 2018-02-01:
Microbee broken in recent versions

It seems lots of targets are broken.
So far I have detected issues in at least these targets:
- VG5K (does not start and reboots)
- MicroBEE (stuck while loading)
- SVI (multiple issues, worse with zsdcc but present on sccz80)
- Z 9001 (no text printed)

It does not seem a problem triggered by optimization because vg5k also fails -O0.

I can confirm that my current code works fine with previous Z99DK versions.

suborb, 2018-02-01:
@Fabrizio-Caruso I just fixed something in the crt code which may be the source of the trouble, if you're not using atexit() then compile with -pragma-define:CLIB_EXIT_STACK_SIZE=0 to disable the atexit stack.

Fabrizio-Caruso, 2018-02-02:
Problem solved now! Thanks!

closed 2018-02-02 by Fabrizio-Caruso