z88dk/z88dk #588: RC2014 / YAZ180 - improve terminal handling - Enhancement Request

feilipu, 2018-02-02:
The character I/O for both these platforms is not perfect,
as standard terminal control codes are not handled.

Character I/O works well when the DCIO build type is implemented,
when working with the BASIC drivers,
but the standard ACIA / ASCI interface is not fully developed.

I need to fix this, and this issue is raised for discussion.

aralbrec, 2018-02-03:
I will have a look at it to make sure it's fixed for the release.

feilipu, 2018-02-05:
A status report on rc2014 as is today.

I'll provide some screenshots for reference, using the password.c example for a standard reference.
Screenshots off a Ubuntu 16.04 with US ASCII keyboard.

I would question the need to maintain the `basic` subtype. I don't understand why the character drivers are sufficient for any purpose, when the terminal driver seem to be necessary to work with the passwork program.

RC2014 has following subtypes
```
SUBTYPE   acia        -startup=0   -Cz"+rom --ihex"
SUBTYPE   basic       -startup=16  -Cz"+rom --ihex"
SUBTYPE   basic_dcio  -startup=17  -Cz"+rom --ihex"
```

YAZ180 has following subtypes

```
SUBTYPE   yabios      -startup=0   -Cz"+rom --ihex"
SUBTYPE   basic       -startup=16  -Cz"+rom --ihex"
SUBTYPE   basic_dcio  -startup=17  -Cz"+rom --ihex"
```

TARGET rc2014
SUBTYPE   asia

The `acia` subtype is for when no `MSBASIC` is available. It is not working currently.
The `acia` build is substantially broken. On boot, it produces `US` as the only output. Three resets appear in below screenshot.

[image]

TARGET rc2014
SUBTYPE   basic
```
zcc +rc2014 -v -subtype=basic -SO3 -clib=sdcc_iy --max-allocs-per-node200000 password.c -o password -create-app
```
This doesn't produce any text on the screen. Pushing random alphanumeric keys finally gets a new line generated. I can't work out which key is producing the new line.

[image]

TARGET rc2014
SUBTYPE   basic_dcio
```
zcc +rc2014 -v -subtype=basic_dcio -SO3 -clib=sdcc_iy --max-allocs-per-node200000 password.c -o password -create-app
```

The backspace character is correct, but the DEL character produces `[3~`.
The cursor (in my case) is generated by the terminal.
`^h` is backspace. `^m` is new line.

[image]

aralbrec, 2018-02-06:
Let's roll #562 into this one.

feilipu, 2018-02-06:
Working on `acia` tonight.

aralbrec, 2018-02-16:
> I would question the need to maintain the basic subtype. I don't understand why the character drivers are sufficient for any purpose, when the terminal driver with its edit buffer seem to be necessary to work with the password program or any program that uses scanf().

It's intended for a terminal on the PC operating in line mode.  In this mode, text is only sent to the rc2014 a line at a time with editing being done on the PC end, for example [putty does this](https://www.ssh.com/ssh/putty/putty-manuals/0.68/Chapter4.html#config-localedit).  The advantage is all the editing features on the PC can be used while the disadvantage is key input is no longer char at a time so video games would be out.

Since basic_dcio is almost working, I'll try to fix that one first.  When you press DEL it looks like an escape sequence is being generated instead of ascii code 127.  There is no escape code interpretation on the input stream but it looks like maybe this has to be added sometime down the road.  The input stream is meant to be plain ascii.

feilipu, 2018-02-16:
AFK till Wednesday. I’ll get the Hex codes for `DEL` sequence then.

aralbrec, 2018-02-16:
If the line mode entry is not useful we should just get rid of it and replace the basic subtype with the basic_dcio type.

The last commit should get rid of the cursor and it adds an opportunity to [reject typed characters](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal/rc_01_input_basic_dcio/rc_01_input_basic_dcio_iterm_msg_reject.asm) to the input driver.  Right now it will reject anything that's not in ascii range [32,126] - and that includes ascii DEL.  If escape sequences have to be rejected, a state machine to interpret them would be implemented here with the flush ioctl resetting the state machine.

Also, the current behaviour is to exit the program if CTRL-C is entered and stdin will close if CTRL-Z is entered.  I'm not sure if that's the most desirable behaviour or not.  Maybe it should be runtime settable behaviour.

I think ^m is [working as intended](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal/rc_01_input_basic_dcio/rc_01_input_basic_dcio_iterm_msg_getc.asm#L29)? ^m is ascii code 13 and this function is translating it to CHAR_LF (\n) for the c side.  So it should act as end of line.

If you agree it's a good idea to get rid of the basic subtype, I'll just replace it with basic_dcio (and call that basic).  I think it may be a good idea to also introduce a tty_z88dk driver on top so that there can be some basic cursor placement and colours done with vt100 escape sequences.

aralbrec, 2018-02-16:
The escape sequence for DEL is `^[3~` according to this link under "PC-Style Function Keys" and also under "VT220-Style Function Keys":

https://invisible-island.net/xterm/ctlseqs/ctlseqs.html

Is there a way to force ASCII input in the terminal settings?

feilipu, 2018-02-16:
Yes I agree killing the line mode `basic` subtype is desirable, for both targets.

And having a `tty` style would be an asset.

For the rom `acia` and `yabios` subtypes, I think they should be best as terminals too. I know the current `yabios` `scanf()` is not quite right, and this may be related to the underlying subtype support?

feilipu, 2018-02-16:
Forcing different codes could work, but AFK so can’t test different terminal options to see, currently. 

But possibly accepting standard PC codes might be an option too?

aralbrec, 2018-02-16:
> For the rom acia and yabios subtypes, I think they should be best as terminals too. I know the current yabios scanf() is not quite right, and this may be related to the underlying subtype support?

Yeah they should all be the same.

> But possibly accepting standard PC codes might be an option too?

I was just wondering how complicated it would be but maybe it isn't that bad.  This is really the prototype for interfacing with modern day terminal programs so doing as complete a job as possible would be ideal.

feilipu, 2018-02-22:
Had a chance to test the [commit](https://github.com/z88dk/z88dk/commit/1cd62cf08b9e6b53acf1f45f99e3686411b55c73).

### Mods added to commit
- add [ACIA Interrupt](https://github.com/z88dk/z88dk/pull/597/files#diff-78e49f687a6bb297ac3e943ff4ea4c05) configuration.
- add `rc_01_input_basic_dcio_iterm_msg_reject` to [rc_01_input_basic_dcio.lst](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal/rc_01_input_basic_dcio.lst)

### Summary
- cursor fixed
- enter key now doesn't work - no response when keyed.
- `^Z` sets off a loop as seen in attached screenshot.
- `^C` does nothing.

[image]

aralbrec, 2018-02-23:
ctrl-z should close stdin so reading using scanf, eg, should return an error.  I think that explains the infinite loop.  But it doesn't explain why the username does not appear after the "Username:" prompt.

You have instances of

```
Username: Password:
"phillip"
""
```

Since "phillip" is in the edit buffer you should see:

```
Username: phillip
Password: 
"phillip"
""
```

The last prompt shows a single letter "P" so individual letters are being echoed to the terminal and not just complete lines.  But you do not see the Username prompt prefilled with "phillip" when the line is edited?

I can see I removed ctrl-c from the [getc routine](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal/rc_01_input_basic_dcio/rc_01_input_basic_dcio_iterm_msg_getc.asm); I must have forgotten about that.

If enter is not working how are you ending a line?  ctrl-m?

aralbrec, 2018-02-23:
I did change the [putc](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal/rc_01_output_basic_dcio/rc_01_output_basic_dcio_oterm_msg_putc.asm#L39) code somewhat when doing crlf.  A quick scan should be made.

feilipu, 2018-02-23:
The two screenshots show separate runs of the program. Both of them show I typed `phillip` and then nothing else. I tried the "Enter" key and no response. Then I tried `^m` and it didn't work either. So I never successfully entered a return within the password program.

That was where I (in the first screenshot) tried `^c` and `^z`. With `^z` the scrolling started, probably related to your comment above.

aralbrec, 2018-02-24:
This is a problem:

```
rc_01_input_basic_dcio_iterm_msg_reject:

   ;   Indicate whether typed character should be rejected.
   ;
   ;   enter:  c = ascii code
   ;    exit:  carry reset indicates the character should be rejected.
   ; can use:  af, bc, de, hl

   ld a,c
   call asm_isprint            ; carry reset if in [32,126]
   
   ccf
   ret
```

It will cause the input driver to reject all control codes < 32 including linefeed, ctrl-c and ctrl-z.  So I'm not sure what happened with that infinite run case but let's see what happens after the next commit.

[ctrl-c is still there](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal/rc_01_input_basic_dcio/rc_01_input_basic_dcio_iterm_msg_interrupt.asm) but I'm not sure if it should be kept or maybe it should be turned into something like ENTER but add a ctrl-c at the end.

aralbrec, 2018-02-24:
I removed ctrl-z which I think is best left for the cpm target only.

feilipu, 2018-02-24:
AFK currently. Will test commits and update this comment with outcomes soonest.

feilipu, 2018-02-27:
Have tested the commits.

It looks like the open issue (from my point of view), is that in both `basic` and `acia` subtypes, a `^c` doesn't end the program in `basic` and `acia`, and `^z` causes the scrolling error discussed above.

I think `basic_dcio` is working as expected.

Perhaps this is the opportunity to remove `basic` subtype, and convert the `acia` subtype to terminal mode?

### `subtype=basic_dcio`

character entry works as desired
backspace works
`^c` exits correctly
other `^~` ctrl characters print a `?` character
`DEL` still produces `[3~`, as expected.

[image]

### `subtype=basic`

character entry works as desired in line mode
no echo provided
line termination with `^m` `^j` (`CR` `LF`)
`^c` doesn't end the program
`^z` sets off the infinite scrolling (closed input issue).

[image]

### `subtype=acia`

character entry works as desired in line mode
no echo provided
line termination with `^m` `^j` (`CR` `LF`)
`^c` doesn't end the program
`^z` sets off the infinite scrolling (closed input issue).

aralbrec, 2018-02-27:
subtype=basic_dcio is gone and replaced by subtype=basic
subtype=acia should behave like subtype=basic now

What's still to do:

* Add a tty_z88dk terminal so the cursor can be moved around.
* Look at eliminating some escape sequences from the input stream.

How does ms basic respond to the del key?

feilipu, 2018-02-28:
> subtype=basic_dcio is gone and replaced by subtype=basic
> subtype=acia should behave like subtype=basic now
>
> subtype=acia has been set to restart on exit by default (eg by a ctrl-c). The rst 0x38 bug should be fixed.
> Let me know if it still works. 

Noted, and both subtypes work as expected. Great! :+1:

> How does ms basic respond to the del key?

MSBASIC responds exactly the same way, with the `[3~` code (`0x5B 0x33 0x7E`).

I will clone the diff to yaz180, and create a PR for you.

aralbrec, 2018-02-28:
> The driver works, but doesn't do line editing correctly.
> Just a quick test with the read_line() in the "shell" in the yabios.

Does the getchar() not return a `\n`?  I will try that out later.

There is a getline() in stdio.h, fyi:

```
#include <stdio.h>

void main(void)
{
   unsigned char *line = 0;
   unsigned int len = 0;
   unsigned int slen;
   
   while (1)
   {
      printf("\nEnter line: ");
      
      fflush(stdin);
      slen = getline(&line, &len, stdin);
      
      printf("\nAllocated length = %u\n", len);
      
      if (slen != 0)
      {
         unsigned char term = line[slen - 1];
         
         printf("Terminating char = %u\n", (int)term);
         printf("\"%.*s\"\n", slen - (term == '\n'), line);
      }
   }
}
```

It also allocates a buffer to read into using malloc/realloc.  `len` will be the total allocated length of `line` which includes the `\n` and a 0 byte to terminate the string.  The returned value `slen` is the length of the actual line read not including the terminating 0.  `slen` might be less than `len` if the buffer is long but a short line is entered.

If `line` is initially zero, getline will malloc the initial before.  If it is not zero, it is assumed the buffer was previously allocated by malloc so that it can be grown with realloc.  In this case, `len` must also reflect the allocated size.

feilipu, 2018-03-01:
> There is a getline() in stdio.h.

Yes I originally implemented (aka [copied](https://brennan.io/2015/01/16/write-a-shell-in-c/)) my shell using `getline()` but when it it didn't return, then I dug one level deeper to `getchar()`.

> Does the getchar() not return a \n? I will try that out later but doing things this way should work.

Originally it didn't work, so I dug one level deeper again.
Probably, based on below comment, it is working now. But I didn't test it again.

I've changed the yabios shell to [use `getline()`](https://github.com/feilipu/yaz180/blob/46404aa63494bfe833004fc84314392689a6caef/yabios/main.c#L1214) and it is working perfectly.

However, there is no local echo. I would have thought that [this terminal flags setting](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/yaz180_rules.inc#L24) would have enabled local echo of the input line, whilst it was being input?

aralbrec, 2018-03-01:
> However, there is no local echo. I would have thought that this terminal flags setting would have enabled local echo of the input line, whilst it was being input?

You've added a separate flags setting for TTY iterms:

```
ifdef(`M4__TTY_ITERM_EDIT_BUFFER_SIZE',,`define(`M4__TTY_ITERM_EDIT_BUFFER_SIZE', 64)')

   IFNDEF TTY_ITERM_EDIT_BUFFER_SIZE
      defc TTY_ITERM_EDIT_BUFFER_SIZE = M4__TTY_ITERM_EDIT_BUFFER_SIZE
   ENDIF
```

zcc must be informed about these special m4 variables that come from pragmas.  The pragmas will cause the file "zcc_opts.def" to be created with the pragma settings and then zcc must read that file to create the m4 variables passed to m4 before the crt is assembled.  So there is an array in [zcc.c](https://github.com/z88dk/z88dk/blob/master/src/zcc/zcc.c#L526) that tells zcc what m4 variables need to be set.

I think if `M4__TTY_ITERM_EDIT_BUFFER_SIZE` is added here, things may work.  I will add this to zcc today.

feilipu, 2018-03-01:
Thanks. Just to note I added the `TTY` configuration after finding echo wasn't working.

They are not connected.

aralbrec, 2018-03-02:
Input terminal flags; your setting looks right:

```
bits 15..10  reserved (includes some state for driver base class)
bit 9 = 1 to enable printed cursor
bit 8 = 1 to enable crlf conversion (will be filtered to generate lf only)
bit 7 = 1 to echo typed input characters
bit 6 = 1 to enable password mode (input printed as *)
bit 5 = 1 to enable line mode (input is edited before being delivered)
bit 4 = 1 to enable cook mode (input characters are interpretted in some way)
bit 3 = 1 for caps lock
bits 2..0  reserved (device state)
```

> I checked password.c and it uses scanf() rather than getline(). I don't know whether this is related to the issue of no echo?

No it shouldn't be.  They all get input from the same place; the routine that does the actual line editing is [console_01_input_proc_getc.asm
](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/drivers/terminal/console_01/input/console_01_input_terminal/console_01_input_proc_getc.asm).

> Edit: I just found this README for the ZX. I don't know whether this is the better / more appropriate way to differentiate between two USARTs?

You can always specify a custom driver instantiation in a compile as shown in that example.  The cpm target has lots of streams specified in its [crt](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/cpm/startup/cpm_crt_4.asm.m4#L68) : stdin, stdout, stderr, stdrdr, stdlst, stdpun.  In most targets the startup value chooses among common configurations and the custom instantiation lets you do something special.

There is a kind of combinatorial explosion that happens as you get more drivers into a target so I'm looking at two things to do about it.  One is to allow driver instantiation to be appended to the crt's instantiation instead of replacing it so that the crt can define stdin,stdout,stderr and the user can easily add more for other uarts, pio, etc.  The other is to introduce a mask that would disable / enable individual file descriptors.  Then a crt could include all drivers but the mask would selectively disable some.

feilipu, 2018-03-02:
Hmm. Ok. Not sure where that leaves the outcome?

I'd like to have two configurable "physical" ports, one for each of the asci USARTs. In addition (as per CP/M example) it would be good to be able to logically assign each of these physical ports to one of the logical streams, as a nice to have.

Is it easiest to add the `TTY` definitions to zcc?
I'll do that in my copy tonight, as I'm testing a ESP01S device on the weekend.

Or would it work better in another way?

Later (possibly in 12 months), I'll have the I2C drivers built in, so that a real display driver, or CRT, or practically infinite other ports or interfaces could be configured. But, let's not get carried away right now.

aralbrec, 2018-03-02:
It's the terminal instantiations that are a problem.
From [yaz180_crt_0.asm.m4](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/startup/yaz180_crt_0.asm.m4#L71):

```
   ; fd 0

   include(`driver/terminal/rc_01_input_asci0.m4')
   m4_rc_01_input_asci0(_stdin, __i_fcntl_fdstruct_0, CRT_ITERM_TERMINAL_FLAGS, M4__CRT_ITERM_EDIT_BUFFER_SIZE)

   ; fd 1

   include(`driver/terminal/rc_01_input_asci1.m4')
   m4_rc_01_input_asci1(_ttyin, __i_fcntl_fdstruct_1, TTY_ITERM_TERMINAL_FLAGS, M4__TTY_ITERM_EDIT_BUFFER_SIZE)

   ; fd 2

   include(`driver/terminal/rc_01_output_asci0.m4')
   m4_rc_01_output_asci0(_stdout, CRT_OTERM_TERMINAL_FLAGS)

   ; fd 3

   include(`driver/terminal/rc_01_output_asci1.m4')
   m4_rc_01_output_asci1(_ttyout, TTY_OTERM_TERMINAL_FLAGS)

   ; fd 4

   include(`../m4_file_dup.m4')
   m4_file_dup(_stderr, 0x80, __i_fcntl_fdstruct_0)
```

The `__i_fcntl_fdstruct_n` are file descriptor structures attached to file descriptor `n`.

The first input terminal instantiation is on file descriptor 0 (normally stdin):

```
m4_rc_01_input_asci0(_stdin, __i_fcntl_fdstruct_0, CRT_ITERM_TERMINAL_FLAGS, M4__CRT_ITERM_EDIT_BUFFER_SIZE)
```

The `__i_fcntl_fdstruct_0` indicates that the input terminal should attach to the output terminal on file descriptor 0.  But that's stdin so the print messages from the line editing will be ignored.  You probably want to attach to file descriptor 2 which is the output half of asci0:

```
m4_rc_01_input_asci0(_stdin, __i_fcntl_fdstruct_2, CRT_ITERM_TERMINAL_FLAGS, M4__CRT_ITERM_EDIT_BUFFER_SIZE)
```

The second input terminal instantiation you probably want attached to the other half of asci1:

```
m4_rc_01_input_asci1(_ttyin, __i_fcntl_fdstruct_3, TTY_ITERM_TERMINAL_FLAGS, M4__TTY_ITERM_EDIT_BUFFER_SIZE)
```

`stderr` is being placed on file descriptor 4, which is a little non-standard as it is normally found on file descriptor 2.  This:

```
m4_file_dup(_stderr, 0x80, __i_fcntl_fdstruct_0)
```

is making `stderr` a dup of file descriptor 0 which is `stdin`.  `stderr` is an output stream so it must be a dedicated output stream or a dup of an output stream.  I'm not sure which output stream you want `stderr` on but maybe the crt one on fd 2.  Then the instantiation should look like:

```
m4_file_dup(_stderr, 0x80, __i_fcntl_fdstruct_2)
```

Besides this, normally `stdin` is on fd=0, `stdout` is on fd=1 and `stderr` is on fd=2.  You don't have to do it that way but then a lot of software assumes this so it's probably best to conform by putting the "main" input and output terminal on fd=0,1,2

aralbrec, 2018-03-02:
> Is it easiest to add the TTY definitions to zcc?
> I'll do that in my copy tonight, as I'm testing a ESP01S device on the weekend.

TTY is a good generic name like CRT.  I think having a few of these that can be used in many targets would be a good way to go.

> Later (possibly in 12 months), I'll have the I2C drivers built in, so that a real display driver, or CRT, or practically infinite other ports or interfaces could be configured. But, let's not get carried away right now.

Yes I will be doing the same with uart and esp in the zx next target soon.  I think I'll add these with the user "append to driver instantiation" option.  The number of crts in the zx next target is going to be silly.  Doing these things now can indicate where there are some shortcomings in the current library drivers.  I already know we need non-blocking io done properly and we need probably a better tty state machine (for escape sequences) on output and a state machine on input to process input escape sequences (this wasn't even considered although the ITERM_MSG_REJECT presents an opportunity to do this).  The big one is the disk io to finally be put in.

feilipu, 2018-03-02:
Updated the [yaz180_crt_0.asm.m4](https://github.com/feilipu/z88dk/blob/4f4bd52cbb7ef5ad8298d8d799ad17c9f45e236c/libsrc/_DEVELOPMENT/target/yaz180/startup/yaz180_crt_0.asm.m4#L69) as suggested, and removed the `zcc.c` related configs. The more I dug into that the deeper I got, and it looked very complex.

It is sufficient to manage the terminal flags separately, until such time as a broader solution is developed.

Anyway. All the line editing works, and remote echo works. :+1:

The only odd thing is that Rx text from `asci1` is being output on `asci0`, even though there is no logical connection present. I will have to check I'm managing the buffers alignment correctly.
[image]

aralbrec, 2018-03-02:
> The only odd thing is that Rx text from asci1 is being output on asci0, even though there is no logical connection present.

I made a quick scan and I didn't see anything.

I added `ttyin` and `ttyout` to yaz180's [stdio.h](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sdcc/stdio.h#L64) so that they are available to C by default (if not present you'll just see a linking error if they are used).

You may want to doublecheck that printing to asci does actually go to asci and not asci0:
`fprintf(ttyout, "Testing asci1\n");`

suborb, 2018-03-03:
> The number of crts in the zx next target is going to be silly. 

Apologies for just picking on one sentence in a very long thread.

I think we've already reached that stage - there's so many permutations and hence CRTs that it's really quite difficult to understand at times.

I know classic is a lot simpler, but I do wonder if we can take something from there - where the startup number defines the general environment (e.g. dot command, ROM, loaded from basic), hence can be aliases with a SUBTYPE. The associated CRT has "sensible" defaults and pragmas can provide tuning behaviour, for example changing the terminal driver from fzx to fixed width.

> One is to allow driver instantiation to be appended to the crt's instantiation instead of replacing it so that the crt can define stdin,stdout,stderr and the user can easily add more for other [...]

The former feels like it would be easier from an end user perspective - fewer magic numbers and hence more descriptive, it would also allow easy addition of 3rd party drivers without needing to incorporate them into the tree and modify the crts.

aralbrec, 2018-03-03:
> I know classic is a lot simpler, but I do wonder if we can take something from there - where the startup number defines the general environment (e.g. dot command, ROM, loaded from basic), hence can [...]

There's a little of that already.  The subtype is being used for output file format (tap, sna, dot, dotx, dotn and at least one more will come that is a single monolithic file).

I added a [startup offset](https://github.com/z88dk/z88dk/blob/master/lib/config/zxn.cfg#L32) so that the subtype can add a constant to the startup number.  In this case, dot commands get 256 added to startup, dotx add 512 and dotn add 768.  The idea is startups 0-31 are choosing standard configurations for stdin/stdout/stderr and the dot subtypes are choosing a new crt with the same stdin/stdout/stderr.  The crts are very different so keeping them separate is better for maintainability.

For startups 0-31 there are several terminal types:

1. 32x24 spectrum res
2. 32x24 spectrum res tty
3. 64x24 spectrum res 4x8 font
4. 64x24 spectrum res tty 4x8 font
5. 32x24 spectrum res fzx
6. 32x24 spectrum res fzx tty
7. 64x24 timex hires
8. 64x24 timex hires tty
9. 128x24 timex hires 4x8 font
10. 64x24 timex hires fzx
11. 63x24 timex hires fzx tty

and I haven't added:

32x24 timex hicolour
64x24 timex hicolour
16x24 ula lores
32x24 ula lores
32x24 layer 2
64x24 layer 2

And the current thinking is one crt per stdin/stdout/stderr for each terminal type with the user custom instantiation used to do mixed combinations.

Maybe one crt would work for this stdin/stdout/stderr selection using something to choose which terminal type is used.

feilipu, 2018-03-03:
> I added `ttyin` and `ttyout` to yaz180's `stdio.h` so that they are available to C by default (if not present you'll just see a linking error if they are used).

Thanks. I also added a `ttyerr` interface in #625. Just to be orthogonal with `stderr`.

> You may want to doublecheck that printing to asci1 does actually go to `asci1` and not `asci0`.
`fprintf(ttyout, "Testing asci1\n");`

I actually just realised that I can't do that testing easily, because I don't have an "app" subtype, although I can easily load an "application" from IDE, or via serial. The existing subtypes are either build the new bios, or use Basic `RST` for IO on one USART.

I need to create a subtype that looks like the yabios subtype, but the actual final device driver calls are pulled from an include file. Everything above `0xF000` in CA1 can be called directly in attached file.
[yabios.inc.txt](https://github.com/z88dk/z88dk/files/1777348/yabios.inc.txt)

The `subtype=app` should originate at `0x0100` like CP/M. 

Related, but off topic, I also have a [`_system_rst`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/driver/yabios/asm_common1_driver.asm#L261) that can be used via a `RST`, and if the destination is in CA1, then it will just bounce it, rather than doing a bank swap.

Does an included file, providing `PUBLIC` labels, have precedence over the library provided labels?

If so the `subtype=app` can be build by simply by adding the file and including the labels from the file. Otherwise I'll have to create a new set of terminal drivers, and mangle up the yabios labels so they don't conflict.

aralbrec, 2018-03-03:
> Does an included file, providing PUBLIC labels, have precedence over the library provided labels?

Yes the linker will use the first label it finds and it will search provided source files before library files.

akalulyadurov, 2018-03-04:
printf() doesn't work correctly. It shows only the first two symbols of printf parameter.
scanf() doesn't work too.

zcc +rc2014 -subtype=acia -clib=sdcc_iy -SO3 -v -m --list --max-allocs-per-node200000 test3.c -o test3 -create-app

**test2.c**

#include <stdio.h>
void main()
{
  	char ch;
	printf("Enter any character \n");
	scanf("%c", &ch);
	printf("Entered character is %c \n", ch);
}

[image]

feilipu, 2018-03-04:
I've just tried to program my RC2014 with
```c
#include <stdio.h>

void main()
{
    char ch;
    printf("Enter any character \n");
    scanf("%c", &ch);
    printf("Entered character is %c \n", ch);
}
```
Using the command line:
```bash
zcc +rc2014 -v -subtype=acia -SO3 -clib=sdcc_iy --max-allocs-per-node200000 test2.c -o test2 -create-app
```
And I get this:
[image]

I have attached the HEX file (remove .txt from name before loading for convenience), so you can see if there is a hardware issue with your RC2014.
[test2.hex](https://github.com/z88dk/z88dk/files/1778295/test2.hex.txt)

Good luck.

akalulyadurov, 2018-03-05:
> If you're using github to update you'll have to rebuild the z80 library.
> From z88dk/libsrc/_DEVELOPMENT:
> Winmake rc2014`` (windows)make TARGET=rc2014``` (other)

Example of Phillip is working correct on my RC2014 mini.
But I don't understand why my .HEX file size isn't the same as test2.hex.
<https://github.com/z88dk/z88dk/files/1778295/test2.hex.txt>

zcc - Frontend for the z88dk Cross-C Compiler - v10912-ae80f23-20180203.
OS: Windows7 32 bit.

Akal

feilipu, 2018-03-05:
Could I suggest opening a new issue please? Your concern is off topic in this thread, and the resolution won’t be easy to find.

Thanks.

I’m AFK for three days (pecking on mobile now). Will look on return.

aralbrec, 2018-03-06:
> But I don't understand why my .HEX file size isn't the same as test2.hex. <https://github.com/z88dk/z88dk/files/1778295/test2.hex.txt>

Your own compile doesn't work?  You'll need to update your z88dk install.

If you used github to clone, an update will update all the source files but you still have to build the binaries and z80 libraries.  For the rc2014, I left some instructions for building the newlib z80 libraries a couple posts up^^.  The binaries (like zcc, etc) don't have to be rebuilt as often because they change more slowly.  You're on windows so you could use vs2015 to build (there is a sln in z88dk/win32).  Or you could copy the z88dk/bin directory from the most recent [nightly build](http://nightly.z88dk.org/).

If you downloaded a nightly build to install, to update just delete or rename the old install and put a new nightly build in the same place.  The environment variables will still be pointing to the right places so it will just work.

If you still can't get a working compile, maybe open another issue.

akalulyadurov, 2018-03-06:
Sorry, It was my mistake. After updating the z88dk just now my .HEX file
became equal to the original .HEX file of Phillip.

Thanks,
Akal

feilipu, 2018-03-07:
From my point of view this work is done.

There are some remaining nice to have items for RC2014 and YAZ180, but these can go in another issue.

### TO DO

- [ ] Add a tty_z88dk terminal so the cursor can be moved around.
- [ ] Look at eliminating some escape sequences from the input stream.

closed 2018-03-07 by feilipu