z88dk/z88dk #589: [SAM][ANSI] After clear screen text displayed with a scroll up + crash if text on last row

Fabrizio-Caruso, 2018-02-04:
The Sam Coupe target  works but has some display issues:

- After a clear screen the text always appears from the bottom up with a scroll (no matter which row the code is trying to display)
- The scroll stops at the last line that is shown at the last row no matter on which row it was supposed to be shown
- Trying to explicitly display something on the last row will produce a crash

suborb, 2018-02-04:
The +sam target uses the ROM print routines.

It looks like the Sam ROM routine 0x014E clears both the upper and lower screen, but then exits with the lower screen selected - hence you get the scrolling up since I guess it's only the lower 2 lines of the screen.

I've changed the cls function to re-select the upper screen so the scroll should go away - at least it does for the ansitest example

That should sort the first issue out, and I'm hoping the others go away too as a result.

suborb, 2018-02-04:
The Sam screen is split into 2 parts - 22x32 upper, and a 2x32 lower.  An attempt to print to row 23 and 24 will result in that error message.

On the Spectrum when using the rst10 printer, there's a way to make the upper screen address all 24 rows, I'm not sure how to that on the Sam unfortunately.

Ideally, we'd do the same as the Spectrum and have a custom print routine but that would have to handle the different screen modes so isn't a trivial addition.

closed 2018-02-04 by suborb