z88dk/z88dk #590: [NASCOM] Keyboard input very slow (maybe new issue) and no auto-repeat (old issue)

Fabrizio-Caruso, 2018-02-04:
It seems to me that the Nascom target has some serious issue with the keyboard input:
- Key detection is SUPER slow (it seems to me much worse than before)
- No-autorepeat: keeping a key-pressed has no effect other than just one single key-press

suborb, 2018-02-04:
I'm just playing around with adv_a and virtualnascom/NASSYS to see what the behaviour is.

Initial key detection seems to be okay (i.e. on a par with NASSYS), but the first key repeat is very slow, subsequent seems ok. This will be using fgetc_cons()/getch() using rst 8 (RIN). This will wait for a key to be pressed. I suspect there might be a poke that can be made to improve key repeat speed - I'll need to look through the scanned nassys documentation to find it.

getk() which just polls the keyboard calls rst 18/0x62 (IN) which doesn't wait. That shows the same behaviour in terms of delays.

Fabrizio-Caruso, 2018-02-04:
I am using the Windows version of Virtual-nascom-20 (which maybe was ported by a friend of ours).

suborb, 2018-02-04:
@zx70 any ideas on the difference behaviour between the Mac build and Windows build? - The Mac one uses SDL.

zx70, 2018-02-07:
Dominic,  I thought at a problem in the emulator but all the emulators I've tried had the same problem. 
your finding halves the problem, because still  nas-sys 3 is the only ROM supporting this type of keyboard scanning.   But I must thank you a lot for the hint!

suborb, 2018-02-07:
So, should we add support for the RKBD driver then?

zx70, 2018-02-08:
I Think so,  most of the nascom computers were not shipped with the nassys3

suborb, 2018-05-29:
Keyboard bits are defined in https://github.com/mamedev/mame/blob/5be75c4a63932afaaad8af9bb9e333d59cf8e788/src/mame/drivers/nascom1.cpp

I think it's read as follows:

out ($00), 2 -> reset keyboard counter
in  a,($00) -> row 0
out ($00), 1
in a,($00) -> row 1
...
etc up to row 8

So we could add an inkey driver hopefully without too much trouble