z88dk/z88dk #592: [CPC] The CPC target produces code that is not compatible with plain CPC (CPC464)

Fabrizio-Caruso, 2018-02-04:
The CPC target produces code that is not compatible with plain CPC (CPC464).

Remark: 
1. I am NOT testing this with any external lib (like crscpc) and not even psg.
2. I am using WinApe 2.0 beta 2 and setting CPC464 (Basic 1.0). You need to restart to make sure the change is taken into account.
3. I am not doing using any Assembly (in the test)

suborb, 2018-02-04:
Sorry, I'm going to have to pass on looking into this one - I'm not on Windows and can't coax any of the emulators into accepting a .wav file.

Fabrizio-Caruso, 2018-02-04:
@suborb , I am on Windows as well and I use WinAPE.
I can create a tape file this way:
```
$(SOURCE_PATH)/../tools/2cdt.exe``` -n -r cross_chase $(BUILD_PATH)/LIGHT_cpc_no_udg.cpc  $(BUILD_PATH)/LIGHT_cpc_no_udg.cdt
```
where the .cpc file is the binary generated by Z88DK.

suborb, 2018-02-04:
@Fabrizio-Caruso thank you, 2cdt is what was needed. A quick compile for the Mac and I can load something into JavaCPC.

It was of course adv_a and it seems to work ok. Can you post an example of something that crashes for you?

Fabrizio-Caruso, 2018-02-04:
@suborb, CROSS CHASE crashes ONLY if you use BASIC 1.0 even the version for CPC that does not use CRSCPCLIB (cpc_no_udg_light). It does work fine with 464PLUS and 6128 models.
My game uses conio/ansi and it is compiled with sccz80. This version of the game does not use any explicit hardware-related code nor Assembly.

make cpc_no_udg_light
will produce the cdt file that can be used with WinAPE.

You can get the precompiled binary in the release section of my GitHub page:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/releases/tag/LotsOfFeatures
File name: LIGHT_cpc_no_udg.cdt

Fabrizio-Caruso, 2018-02-04:
@suborb, I am able to reproduce the problem ONLY with WinAPE but I do not get any crash with JavaCPC.
Maybe we need someone who could test on the real hardware...

Fabrizio-Caruso, 2018-02-05:
I wonder which emulator is more accurate...

Fabrizio-Caruso, 2018-02-06:
Bad news: MAME confirms the bug that WinAPE shows. I cannot reproduce it under JavaCPC.
I am assuming that MAME and WinAPE are more accurate than JavaCPC. It would be weird that two completely different emulators fail at the exact same place.

Fabrizio-Caruso, 2018-02-06:
A simple hello world crashes the CPC target if run under a plain CPC464:
```
#include <stdio.h>
  int main() {
    printf("Hello world!\n");	
    return 0;
  }
```
compiled with
```
cpc_hello:
	$(Z88DK_PATH)$(MYZ88DK) +cpc -O3 $(SOURCE_PATH)/../experiments/cpc_hello.c \
	-clib=ansi  -lndos -create-app -o $(BUILD_PATH)/cpc_hello.prg
	$(SOURCE_PATH)/../tools/2cdt.exe -n -r cross_chase $(BUILD_PATH)/cpc_hello.cpc $(BUILD_PATH)/cpc_hello.cdt
```

suborb, 2018-02-06:
A thing to notice is that even when running hello world in JavaCPC on the 464, there's some screen corruption going on that looks like it's stack pointer related.

Compiling with -pragma-define:REGISTER_SP=-1 will prevent the crt picking up the stack from (0xae60) and allow hello world to work in Mame (and hopefully in other emulators as well)

It looks like we decided on *0xae60 since that's meant to be the address of the UDGs (See https://www.z88dk.org/forum/viewtopic.php?pid=13842) but maybe there's a different layout in play depending on whether AMSDOS is loaded on the 464.

Fabrizio-Caruso, 2018-02-06:
@suborb, even CROSS CHASE seems to WORK!!! :-)
Need to test it more and under all configurations!

suborb, 2018-02-06:
Ah, looking at the CPC system variables here: http://www.cantrell.org.uk/david/tech/cpc/cpc-firmware/mem-use.htm it looks like 0xae60 isn't available on the CPC464, only on the 6128. That would explain the problems.

I've not tried it, but for the CPC464 you might need to compile with -pragma-define:REGISTER_SP=-0xAE7D to pick it up from its version of the system variable.

Fabrizio-Caruso, 2018-02-06:
@suborb, -pragma-define:REGISTER_SP=-1 seems to work.
I need more time and tests to tell you for sure.

Fabrizio-Caruso, 2018-02-07:
@suborb, -pragma-define:REGISTER_SP=-1 seems to work on ALL configurations.
Is this something surprising? Should I re-test it with different emulators?

suborb, 2018-02-07:
It makes sense - reading the docs suggests that the CPC has a 256 byte stack area reserved in a safe area which is probably what we're using in this case. As long as your stack frames don't get too large it'll work.

suborb, 2018-02-07:
I've updated the wiki (https://www.z88dk.org/wiki/doku.php?id=platform:amstradcpc) with all that we've learnt.

Ideally we'd autodetect the machine and adjust sp accordingly, but I'll leave it as is for the moment.

closed 2018-02-07 by suborb

cpcitor, 2018-02-08:
Hello. Do I understand correctly that z88dk default settings compiles to something that does not work on all CPC models? That feels wrong and will probably bite a number of people who just try z88dk. 

AFAIK most projects of the era, and most project nowadays just go fine with the default stack size.

Is there a reason to believe a deep stack is necessary by default, eating more precious bytes and wasting default stack area?  Else does it make sense to lose compatibility between CPC models just for the sake of a non-necessary deep stack default settings?

IMHO only projects specifically needing a deep stack should need to care, not the other way around.
Then default compilation settings become naturally compatible with all CPC models.

What do you think?

suborb, 2018-02-08:
Thanks for the confirmation - I don't think we knew that most 3rd party apps just used the default stack area. And for confirming that that is what is happening with the setting.

Switching over is pretty easy and I'll do that now that I know what's happening.

suborb, 2018-02-12:
In e7bbebc0d6f5b6fbfc2ac6dbe8cadfec8878347a I removed the explicit picking of sp. This can still be done manually as necessary.