z88dk/z88dk #593: copt: crashes for non-ascii input

aralbrec, 2018-02-05:
I thought this was an open issue already but I couldn't find it.

copt will crash if the input file contains non-ascii characters.  The specific report involves input with cyrillic characters in comments.

aralbrec, 2018-02-05:
I'm not sure if the solution is as simple as opening the input file in binary mode so no processing of the text stream is done by stdio.

suborb, 2018-02-05:
I'll fix this. This is due to this code in the install function:

        register char *s;
        int  i;
        for (i = 0; *s; i += *s++)
                ;
        i %= HSIZE;

       struct node *p = htab[i];

Where i ends up being a negative index if *s >= 128 which will be the case for there's a non ASCII utf8 character in there.

commit ab8cbecf (2018-02-05): Fix #593: Handle non-ascii text correctly
closed 2018-02-05 by suborb