z88dk/z88dk #596: YAZ180 - support yabios inline RST system calls - enhancement request

feilipu, 2018-02-06:
Related to #595 the current mechanism for in-line assembly RST instructions as[ macros in a header](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/yaz180.h#L69) file doesn't work. It won't compile.

```c
#define call_error(code)        \
    do{                         \
        __asm                   \
        rst 8H                  \
        defb code               \
        __endasm;               \
    }while(0)

#define call_far(addr, bank)    \
    do{                         \
        __asm                   \
        rst 10H                 \
        defw addr               \
        defb bank               \
        __endasm;               \
    }while(0)

#define call_sys(addr)          \
    do{                         \
        __asm                   \
        rst 20H                 \
        defw addr               \
        __endasm;               \
    }while(0)

#define call_apu(cmd)           \
    do{                         \
        __asm                   \
        rst 28H                 \
        defb cmd                \
        __endasm;               \
    }while(0)
```

A complete solution was developed for [`break()`](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/yaz180.h#L104) in #374 but that was [quite complicated](https://github.com/z88dk/z88dk/blob/master/lib/yaz180_rules.1#L2) and fragile because of the various SDCC optimisation techniques.

Ideally, a mechanism of inserting a non-optimising but byte counting mechanism for in-line assembly is needed. Building this the same way that `break()` was built is probably not the answer. Probably the solution to this and #595 would be similar.

suborb, 2018-03-10:
Just picking this up (a bit late).

This was something that I was experimenting with a long time with sccz80 - I was thinking about implementing shared libraries on the z88 and could hijack a rst to call via.

As a result,  `__SHARED__` and `__SHAREDC__` function decorators were introduced.`__SHARED__` just did a call [function]_sl (which would be a stub to a rst), and `__SHAREDC__` did a rst 8/defw [funcname]

Of course, calling via a rst introduced a stack offset, such that arguments weren't at sp+2 on entry to the function which is where the -shareoffset came in which added a displacement when compiling them.

Of course this is all a bit dusty and irrelevant since I don't think sdcc has similar concepts and keeping source compatibility is important these days.

suborb, 2018-03-10:
The attached diff simplement __z88dk_shortcall(rst_number, value) for both compilers. It's a rough cut and doesn't consider a call through a function pointer.

I'm not sure what side effects will come out in zsdcc, but it might be worth trying to see if they sort out the underlying issues and then they can be finessed.

[sccz80_shortcall.diff.txt](https://github.com/z88dk/z88dk/files/1799888/sccz80_shortcall.diff.txt)
[sdcc_shortcall.diff.txt](https://github.com/z88dk/z88dk/files/1799889/sdcc_shortcall.diff.txt)

feilipu, 2018-03-11:
Thanks. I will look at what happens with the generated code.h

The fix noted in #637 using the `RST` functions is not really a complete solution. I think your idea around the shared libraries feature is much more in line with what I think is needed for bankable platforms. In my case it comes down to two needs.
1. reusing library code that exists in `BANK_0` using , and 
2. allowing application code to call across banks.

Jack Ganssle did a [review of the (now old) compilers](http://www.ganssle.com/articles/ammu.htm) and how they handled banking. Unfortunately none of these compilers have been released, and Softools still require substantial [$$$ to acquire](http://www.softools.com/scz180.htm) their compiler.

The need can be generalised to any size and location of bank, but for the YAZ180 I have decided to have simple 64kB banks, aligned with the `BBR` register, for simplicity in calculating the far address.

Also, to further simplify the problem, I've tried to limit the `call_sys()` and `call_far()` functionality to register passing functions. Stack passing is another huge increment on complexity.

Example of how it might look:

```c
   //FRESULT f_opendir (DIR* dp,const TCHAR* path);                   /* Open a directory */
extern FRESULT f_opendir(DIR* dp,const TCHAR* path);

    call_sys( f_opendir(DIR* dp,const TCHAR* path) ); // call the system library in BANK_0
    call_far( f_my_opendir(DIR* dp,const TCHAR* path), -1); // call one bank lower

// OR WITH DECORATORS

    f_opendir(DIR* dp,const TCHAR* path) __SHARED__; // call the system library in BANK_0
    f_my_opendir(DIR* dp,const TCHAR* path), __BANKED-1__); // call one bank lower

```
The problem with the `RST` function macros in #637 is that they don't account for loading the registers required for the destination function before making the call via the `RST` mechanism, nor does it allow the compiler to understand that a return value is also available after the macro `RST` has completed.

The `RST` macros are just handy 1 byte trampolines, that allow passing of a destination address and a bank (taken together as a far pointer), without impacting the main register contents. Or, in the case of `_call_error()`, to allow a parameter to be passed without affecting register or stack contents.

Is there an existing C library capability to load/unload a specific register with a specific variable?

This would make the solution as it stands quite workable, if not very tidy.

suborb, 2018-03-11:
Stack based parameters across banks will work with sccz80 - the -share-offset= parameter will offset the parameters by as much as is needed by the trampoline. I think we could contrive a macro syntax that will work with sdcc as well (by offsetting ix after entry using a __asm block). The downside is obviously that a call from the same bank to that function won't work.

However for a library like functionality that might be okay (eg there's a local library entry point at a known address, and the implementation elsewhere within the bank) - it certainly allows forwards compatibility in terms of changing code size.

If it comes to it, then these stubs could be written manually - all they'd be doing would be gathering parameters from higher up the stack and re-pushing them then calling the real implementation.

feilipu, 2019-02-25:
This is also related to the general case in #241

feilipu, 2019-12-08:
Work around [here](https://github.com/z88dk/z88dk/commit/ea82cd55098a10d70f5f8764d9a2ecbd1ff25b4d).

closed 2019-12-08 by feilipu