z88dk/z88dk #598: [Z1013] Not working (hello world tested with J1013)

Fabrizio-Caruso, 2018-02-07:
The Z1013 seems to not produce anything that runs.

I have tested it with both my game and a simple hello world.
```
#include <stdio.h>
  int main() {
    printf("Hello world!\n");	
    return 0;
  }
```
compiled with
``` 
z1013_hello:
	$(Z88DK_PATH)$(MYZ88DK) +z1013 -O3 $(SOURCE_PATH)/../experiments/hello.c \
	-lndos -vn \
	-create-app -o $(BUILD_PATH)/z1013_hello.z80
``` 
tested with J1013 by following the instructions (Starten)

suborb, 2018-02-07:
Following exit of a program, it looks like the z1013 clears the screen and resets back to the prompt.

I think that's what's happening with this hello world example - just loop or print more lines and you'll see the output.

However, I can see that the non-ANSI version isn't doing the required lf/cr conversion, so I've fixed that up in df77a4699cce1d36a5cb4600eebd288e093e43bf

Fabrizio-Caruso, 2018-02-08:
@suborb, How do you run the hello world example?
I have added a while(1){} statement but under J1013 emulator
Datei->Laden->Select .z80 file->Starten has no effect
i.e., I see the prompt.

Fabrizio-Caruso, 2018-02-08:
@suborb, I confirm with a recent version of Z88DK, that the printf("hello world\n") followed while(1){} produces no visible effect when loaded following the  instructions found in the Z88DK documentation.
(Datei->Laden->Select .z80 file->Starten). Am I missing some important command to start the program?
Am I compiling it with the wrong options?

suborb, 2018-02-08:
That’s how I’m compiling and loading it - it loads and runs automatically. 

Dumb question but you have selected the right machine to emulate? Spanner - z1013 - 64k option (bottom) - top button on right.

Fabrizio-Caruso, 2018-02-08:
No dumb questions.
There is no Option button in J1013 which only emulates Z1013 as far as I can tell

Fabrizio-Caruso, 2018-02-08:
@suborb, could you please tell which emulator you are using?
I am using the two emulators suggested by the Z88DK documentations:
- J1013 Version 2.2 (latest), which only emulates Z1013 (http://www.jens-mueller.org/j1013/download.html)
- KCemu 5.1 (I select Z1013 at start-up)
In both cases, if I load/run, nothing happens.

suborb, 2018-02-08:
I was using jkcemu (http://www.jens-mueller.org/jkcemu/download.html) but I've just tried adv_a out with j1013 and that works too

[image]

Fabrizio-Caruso, 2018-02-08:
Weird...
Maybe the Windows versions of the emulators are broken...
I have tested this on both Windows 10 and Windows 7.
I am using a recent version of Z88DK and recent versions of the most emulators mentioned above.
I am compiling as shown above
```
z1013_hello:
	$(Z88DK_PATH)$(MYZ88DK) +z1013 -O3 $(SOURCE_PATH)/../experiments/hello.c \
	-lndos -vn \
	-create-app -o $(BUILD_PATH)/z1013_hello.z80
```
I am loading/running the .z80 and I just see the prompt and nothing else.

Fabrizio-Caruso, 2018-02-08:
@suborb, which extension do you give to the binary file? I am using .z80 as suggested by the doc.

suborb, 2018-02-08:
Ah. I've just tried with java9 and I get the same problem as you - stuck at the prompt.

When it works I'm using java8 (1.8.0_131)

Fabrizio-Caruso, 2018-02-08:
Just a sad prompt:

[image]

Fabrizio-Caruso, 2018-02-08:
@suborb, thanks a LOT for funding the issue!
Sorrt for bothering you!! I have tried many things before asking... but not the good ones...

Fabrizio-Caruso, 2018-02-08:
@suborb, what sucks is that all emulators are broken under Java 9...
I hope Mame/Mess emulates the Z1013.

suborb, 2018-02-08:
Actually, that may not be it, I clicked Laden instead of Starten.

That's a weird initial prompt that you've got there, from a fresh start mine looks like this:

[image]

And then after selecting the file I get these parameters:

[image]

[image]

Fabrizio-Caruso, 2018-02-08:
I did not post the inital one. I posted what I get after Starten.

Fabrizio-Caruso, 2018-02-08:
This is what I see after selecting the .z80 file

[image]

Fabrizio-Caruso, 2018-02-08:
After that I click on Starten and I what I get is just a second prompt as display in my initial image.

Fabrizio-Caruso, 2018-02-08:
My JAVA version is:
java version "1.8.0_121"
Java(TM) SE Runtime Environment (build 1.8.0_121-b13)
Java HotSpot(TM) Client VM (build 25.121-b13, mixed mode, sharing)

suborb, 2018-02-08:
I love the description! 

Without my fix from last night I do get a weird display with just hello world (and adv_a is full of blocks). -clib=ansi isn't dependent on it though, so maybe try that if you've not got that change.

suborb, 2018-02-08:
But then again you said the cross chase doesn't do anything either.

Can you share the output binary that you're creating (so a.bin) - I'll check that it matches up with what I'm creating.

Fabrizio-Caruso, 2018-02-08:
-clib=ansi does not change the situation...

```
#include <stdio.h>

  int main()
  {

	printf("Hello world!\n");	
	
	while(1){};
	
    return 0;
  }
```
compiled with: 
```
z1013_hello:
	$(Z88DK_PATH)$(MYZ88DK) +z1013 -O3 $(SOURCE_PATH)/../experiments/hello.c \
	-lndos -vn -clib=ansi \
	-create-app -o $(BUILD_PATH)/z1013_hello.z80
```

Rename a.zip to a.bin
[a.zip](https://github.com/z88dk/z88dk/files/1706660/a.zip)

suborb, 2018-02-08:
Ah, well, that explains it nicely:

```
hexdump -C a.zip
00000000  00 01 8e 0f 00 01 00 00  00 00 00 00 43 d3 d3 d3  |............C...|
00000010  62 75 69 6c 64 2f 7a 31  30 31 33 5f 68 65 6c 6c  |build/z1013_hell|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000ea0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff        |..............|
00000eae
```

There's no program being loaded into the .z80 file! Well, apart from a load of rst $38

I'll look at appmake later, but I think the problem might be in the make target, try -o $(BUILD_PATH)/z1013_hello.bin - I've got a hunch that the assembler is creating the .z80, then appmake is truncating it to create the output .z80

Fabrizio-Caruso, 2018-02-08:
@suborb , thanks A LOT.
I can now build a correct binary. I need to do:
```
z1013_hello:
	$(Z88DK_PATH)$(MYZ88DK) +z1013 -O3 $(SOURCE_PATH)/../experiments/hello.c \
	-lndos -vn -clib=ansi \
	-create-app -o
	mv $(BUILD_PATH)/../A.Z80 $(BUILD_PATH)/z1013_hello.z80
	rm $(BUILD_PATH)/../a.bin
```

suborb, 2018-02-08:
That's great, I'm sorry we went around the houses to get there.

closed 2018-02-08 by suborb

Fabrizio-Caruso, 2018-02-08:
@suborb thanks and sorry for not figuring it out myself!