z88dk/z88dk #599: [Z1013] kbhit and getk lock the computer with example

Fabrizio-Caruso, 2018-02-08:
It seems that kbhit and/or getk are broken in Z1013.
getk is waiting for keyboard input.

The following code is supposed to 
```
#include <stdio.h>
#include <conio.h>

int main()
{
	printf("Press key\n");

	while(kbhit())
		getch();
	while(!kbhit())
	{ 
	}; 
	getch();

	printf("key pressed\n");
	
	return 0;
}
```
wait for a key press and display the effect of the key press.

If compiled compiled for Z1013 with
```
z1013_wait_press:
	$(Z88DK_PATH)$(MYZ88DK) +z1013  $(SOURCE_PATH)/../experiments/wait_press.c \
	-lndos -vn -clib=ansi \
	-create-app -o
	mv $(BUILD_PATH)/../A.Z80 $(BUILD_PATH)/z1013_wait_press.z80
	rm $(BUILD_PATH)/../a.bin	
```
it will lock forever.

suborb, 2018-02-08:
kbhit() is a wrapper around getk() so that's to be expected. 

I'll look into this after the weekend.

suborb, 2018-02-12:
I've changed getk() to call INKEY (rst $20/defb 4) instead of INCH. INKEY should just poll the keyboard.

However, the behaviour seems a bit odd with the ROM call only returning when the key is released?

Fabrizio-Caruso, 2018-02-17:
@suborb you changes have some positive effects: the game is sort of playable but keeping a key pressed will "pause" the game. 
Not perfect but this is a significant improvement.

Fabrizio-Caruso, 2018-03-20:
@zx70 in my humble opinion this issue is more of a bug than an enhancement.
The latest behavior I could test was that the program is pause when a key is kept pressed.
It is better than nothing and slightly better than the previous behavior (pause when no key is pressed) but it is not behaving as expected.
It is not high priority because it does not prevent the target to be usable.

suborb, 2018-03-20:
There's not much that can be done in terms of calling a different bios routine - we've tried both of them and they're really rather unsatisfactory.

We'd need to write a (userland) keyboard driver to address that issue - assuming that it is software and not hardware that's causing the issue.

zx70, 2018-03-27:
even worse,  it is not guaranteed that this new bios routine would exist on all the z1013 models, I'm attempting to detect the bios version and do some extra magic where possible (relying on the lastk sysvar)

zx70, 2018-04-04:
The fix seems to work

Fabrizio-Caruso, 2018-04-04:
This very simple test program won't work (tested with snapshot 20180404, J1013 with monitor 2.2):
```
#include <stdio.h>
#include <conio.h>
int main()
{
	int i;
	for(i=0;i<10;++i)
	{
		printf("Press key\n");
		while(kbhit())
			getch();
		while(!kbhit())
		{ 
		}; 
		getch();
		printf("Key pressed: %d\n\n", i);
	}
	return 0;
}
```

Fabrizio-Caruso, 2018-04-04:
@suborb 
The result is just the display of "Press key". Then it does not respond to any key press.
I have used the J1013 emulator (which starts wih 2.2 by default).
The binary is compiled with a code above with the following commands:
```
z1013_getk:
	$(Z88DK_PATH)$(MYZ88DK) +z1013 \
	-o $(BUILD_PATH)/z1013_getk.bin -vn -lndos \
	$(SOURCE_PATH)/../experiments/wait_press.c	
```

zx70, 2018-04-06:
Fabrizio, please try again tomorrow.
The change introduced by Dominic (which I broke lately) was already valid for Monitor A.2 (Alphatastatur).
I now added a fix for Monitor 2.02 (Flat Foil Keyboard), which the default configuration on the emulators, it is not perfect (number keys need to be released and pressed again after a while) but much better.
Other Monitors are still hanging the program loop when a key is kept pressed.

zx70, 2018-04-06:
Same fix just applied everywhere (all the monitor versions), they simply liked it and work better.
The only exception are the numeric keys on the flat foil keyboard, where getk stops receiving the ascii code when the key is kept pressed for a while.

closed 2018-04-06 by zx70

Fabrizio-Caruso, 2018-04-07:
I confirm that it is working as it should now! :-) Great work!