z88dk/z88dk #601: [INFO][Z88]How load into an emulator?

Fabrizio-Caruso, 2018-02-09:
Hi

In the doc I read that by default the produced binary can be loaded by BBC BASIC, i.e.,
just compiling with "zcc +z88 ....." without "-create-app...".

Which emulator can be used? Which commands I need to run on the emulator?
Is OZvm ok? 
Do I need to convert the binary into a tap or something that the emulator can digest?

suborb, 2018-02-09:
An application is easiest to load into ozvm. 

The basic target is really only useful on a proper machine - basically for me when I had no way to erase EPROMs and ram installer wasn’t available.

Fabrizio-Caruso, 2018-02-09:
Thanks @suborb ! Where can find the instructions on how to load it as applucation?

Is there a problem in the doc?
"By default, z88dk generates binaries that can be LOADed into the BBC Basic application and run. Using this target is the easiest way to get an application up and running on the z88."

suborb, 2018-02-09:
For a real z88 it is - just xmodem it across. 

For an emulator creating an app is easier. 

Compile with -subtype=app -create-app and in ozvm click on one of the card slots and load the .63 file. The vm will soft reset and your app will appear in the menu. 

There’s a few examples of apps in the usual place.

Fabrizio-Caruso, 2018-02-09:
@suborb, thanks!

Fabrizio-Caruso, 2018-02-10:
@suborb, if I follow your instructions I get:
```
/cygdrive/c/z88dk/bin/zcc.exe +z88 \
-D__Z88__ \
-DTINY_GAME \
-subtype=app -create-app \
-vn -lndos \
src/display_macros.c src/enemy.c \
src/level.c src/character.c src/text.c  \
src/strategy.c src/input_macros.c src/main.c
Error at file 'c:/z88dk//lib/target/z88/classic/z88a_crt0.asm' line 131: symbol 'applname' not defined
1 errors occurred during assembly
Errors in source file c:/z88dk/\lib\target\z88\classic\z88_crt0.asm:
Error at file 'c:/z88dk//lib/target/z88/classic/z88a_crt0.asm' line 131: symbol 'applname' not defined
                   ^ ----         ld      hl,applname   ;Name application
make: *** [Makefile:1119: z88_tiny] Error 1
```

Fabrizio-Caruso, 2018-02-10:
@suborb, if instead I follow the doc and use
	-create-app -make-app 
I get
Unknown option -make-app

Fabrizio-Caruso, 2018-02-10:
Can I compile a program as is if I want to make an application or whatever is necessary?
I would like to have my game run under an emulated Z88 without having to change its code.
Is it possible?
Or do I need to make a special program out of it?

Fabrizio-Caruso, 2018-02-10:
I am closing this issue. What I have read in the doc implies that I have to re-code the game.

closed 2018-02-10 by Fabrizio-Caruso

suborb, 2018-02-10:
It’s been a while since I wrote an app and I’m computer-less at the moment but I think you just need a couple of things to interface nicely:

The application dor which defines the application name, hot key and memory requirements - take a look at the examples to see how that is done - z88dk will work out the memory requirements for you!

A handlecmds function which will allow the os to quit the app. Again the examples have that in. 

You can get away without the redraw functionality if you mark your app as ugly in the application dor.

These things are needed because the z88 runs an is that can cope with several applications running at the same time. On the z88 taking over the whole machine simply isn’t an option. 

That sounds scary and limiting but it’s not. In the past I ported a number of games to the z88 and apart from the few bits needed above the code was pretty much identical.

Fabrizio-Caruso, 2018-02-10:
@suborb thanks for replying and the support! It is nice of you!
The strict rule of my crazy project is to use the same code for all 8 bit computers/consoles.
I will come back to the z88 target later. Thanks a lot!

Fabrizio-Caruso, 2018-09-18:
I am also trying with ZEsarus, which has an explicit menu to load .63 files.
I select the type "eprom".
Zesarus complains about the size of a.63 (16k). It expects either 32k, 128k or 256k.
This is the same in OZVM. There is no 16k eprom format. OZVM displays "fail".

So there may be a problem with appmake.
The a.63 file is too small for any emulator and probably also for the real computer.
Maybe it just needs some padding. I have no idea.