z88dk/z88dk #604: Confused why zsdcc is not picking up my `typedef`s

drako0812, 2018-02-13:
I'm getting a `syntax error: token -> 'WVRN_DWORD' ; column 14` when trying to compile the following code:

`disk.c`
~~~.c
#include <z80.h>
#include "disk.h"
// A bunch of defines
WVRN_DWORD wvrn_disk_size() {
    z80_outp(WVRN_DISK_COM_PORT, WVRN_DISK_REQ_SIZE);
    WVRN_DWORD sz1, sz2, sz3, sz4;
    sz1 = z80_inp(WVRN_DISK_INP_PORT_1);
    sz2 = z80_inp(WVRN_DISK_INP_PORT_2);
    sz3 = z80_inp(WVRN_DISK_INP_PORT_3);
    sz4 = z80_inp(WVRN_DISK_INP_PORT_4);
    return (sz1 << 24) | (sz2 << 16) | (sz3 << 8) | sz4;
}
// Other functions with similar issues
~~~

`disk.h`
~~~.c
// Include Guard
#include "../wvrn.h"
// Some defines and function prototypes
~~~

`../wvrn.h`
~~~.c
// Include Guard
#include <stdint.h>

typedef uint8_t WVRN_BYTE;
typedef uint16_t WVRN_WORD;
typedef uint32_t WVRN_DWORD;
typedef int8_t WVRN_S_BYTE;
typedef int16_t WVRN_S_WORD;
typedef int32_t WVRN_S_DWORD;
typedef uint8_t WVRN_BOOL;
typedef uint8_t WVRN_BYTE;

#define WVRN_TRUE ((WVRN_BOOL)0x01)
#define WVRN_FALSE ((WVRN_BOOL)0x00)
~~~

Am I missing something obvious? sccz80 was able to compile this.
Also for zsdcc (in another file from the same project) I had to move
a variable declaration one scope up from where I had it. Specifically, it
was a variable declared within the scope of a loop. Seems like a similar
issue if I'm not mistaken.

suborb, 2018-02-14:
I’m not run this but have tried moving your variable definition above the z80_outp? Sccz80 is now quite liberal regarding variable placement but zsdcc might not be so.

aralbrec, 2018-02-14:
Both compilers require variables to be declared at the top of a block.

So this is ok:

[code]
..
{
    WVRN_DWORD sz1, sz2, sz3, sz4; // LINE REFERENCED IN THE ERROR
    z80_outp(WVRN_DISK_COM_PORT, WVRN_DISK_REQ_SIZE);
    {
        WVRN_DWORD sz1, sz2, sz3, sz4; // LINE REFERENCED IN THE ERROR
        z80_outp(WVRN_DISK_COM_PORT, WVRN_DISK_REQ_SIZE);
    }
}
[/code]

But this is not:

[code]
..
{
    z80_outp(WVRN_DISK_COM_PORT, WVRN_DISK_REQ_SIZE);
    WVRN_DWORD sz1, sz2, sz3, sz4; // LINE REFERENCED IN THE ERROR
    {
        z80_outp(WVRN_DISK_COM_PORT, WVRN_DISK_REQ_SIZE);
        WVRN_DWORD sz1, sz2, sz3, sz4; // LINE REFERENCED IN THE ERROR
    }
}
[/code]

because here you have a function call ahead of variable declarations in a block {..}.  I used an unusual example of a block nested in another one to emphasize that variable declarations can appear at the head of any block and not just the top of a function.

The ability to declare variables at places other than the top of a block is a C99 feature IIRC that neither compiler fully implements.  The only real exception is that both compilers will now accept declarations inside the initializer of a for loop (for sccz80 this is in recent builds only):

[code]
for (unsigned char i = 0; i != 100; ++i)
...
[/code]

drako0812, 2018-02-14:
Yup, that fixed it. Thank you. For sdcc having supposedly better standards conformance, it seems like a strange problem. Unless, that is more standards-compliant, I'm not sure I'm more used to C++.

closed 2018-02-14 by drako0812

aralbrec, 2018-02-14:
Just FYI, you can get both compilers to inline io instructions if you declare the io port(s) as sfrs (special function registers):

```
__sfr __at 0xfe IO_FE;   // this is an 8-bit io port
__sfr __banked __at 0xfffd IO_FFFD;   // this is a 16-bit io port

// write to port

IO_FE = WVRN_DISK_REQ_SIZE;   // inlines 8-bit io instruction like "out (0xfe),a"
IO_FFFD = WVRN_DISK_REQ_SIZE;  // inlines 16-bit io instruction like "ld bc,0xfffd; out (c),a"

// read from port

x = IO_FE;   // inlines 8-bit io instruction like "in a,(0xfe)"
x = IO_FFFD;   // inlines 16- bit io instruction like "ld bc,0xfffd; in a,(c)"
```

z80_outp() will work fine but it is a function call.

suborb, 2018-02-14:
To follow up, I've just confirmed that sccz80 supports variable declaration that isn't at the top of a block.

feilipu, 2018-02-14:
@aralbrec is the `__sfr` an option to simplify the `BREAK()` in-lining issue I was facing with yaz180 single step command?

I think it might have been another possibility simpler solution? Thoughts?

There are a few other registers for other devices I could do this with too, if it is a suitable robust solution.

aralbrec, 2018-02-14:
>  is the __sfr an option to simplify the BREAK() in-lining issue I was facing with yaz180 single step command?

No I don't think so.  `__sfr` is part of the compiler's grammar so adding new symbols for inlined code using the same method would mean adding them to the grammar and the code generator which is not what you'd want because potentially you have dozens of targets with their own conventions or needs.  Unless I've misunderstood?

The BREAK-inlining thing is similar to how the built-in [call-to-rst inlining](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_opt.9#L10), which affects functions with names of a certain form, and the [intrinsics](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_peeph.0#L192) are working.  A header tells the compiler that a function affects certain registers so that the code generator can generate code around it and then the peepholer replaces the function call with inlined code.  But I think all this can be replaced by #595.

#595 will have the final peephole set used by zsdcc constructed by zcc by piecing together architecture-specific rules, tool-supplied rules and optional user-supplied rules before the compile process starts.  The thought I have in my head is that the rules supplied by the toolset will be assigned priority numbers spaced out and the user will be able to insert his rules at the correct precedence by supplying a priority number that places it among the ones being supplied.  The architecture rules would be responsible for standard code inlining and because it's in zsdcc's peepholer, we can also have the inlined code (somewhat) locally optimized at the same time.  sccz80 doesn't have the problems that need to be solved for zsdcc.

feilipu, 2018-02-14:
Yes you’re right for the #596 `RST` code generation issue. It’s the same as #595, too. 

But I think the `BREAK()` case specifically is just [inserting a 16 bit IO in-line](https://github.com/z88dk/z88dk/blob/master/lib/yaz180_rules.1). So probably using `__sfr` would apply, and simplify things.

AFK currently, otherwise I’d be able to experiment and report.

aralbrec, 2018-02-14:
You're right, those can be replaced with sfrs.  I was looking at the rst ones only.