z88dk/z88dk #605: [X1] keyboard input either crashes (X1 millenium) or freezes/locks (Takeda x1.exe)

Fabrizio-Caruso, 2018-02-15:
It seems that something is really wrong in either kbhit and/or getk in the X1 targets.

Under X1 Millenium emulator I get an immediate crash after pressing a key.

Under x1.exe by Takeda the first key press is detected but then emulated computer locks/freezes.

suborb, 2018-02-15:
Some notes:

I don't think the problem is with getk as such - all that does is read some memory.

However, keyboard scanning is performed by an interrupt routine in the crt. I've no idea (as yet) how the keyboard ports are hooked up so I don't fully understand what's going on, but it does look like a busy loop on the interrupt is possible. This might explain the freeze

suborb, 2018-02-15:
I've found the source code for S-OS running on the X1 from http://www.retropc.net/ohishi/s-os/SWTCX1.zip and it looks like the keyboard scanning is slightly different there.

I'll need to obtain a working emulator for the Mac before I can look any further.

Fabrizio-Caruso, 2018-02-15:
I could find an old emulator:
https://emulationrealm.net/downloads/file/2718-x-millennnium-mac

Maybe worth a try. "X-Millenium" is open source. With some luck, it may be ported or simply compiled for the Mac.

Obviously you can run a Windows emulator under Wine (with some luck).

suborb, 2018-02-15:
That emulator is compiled for PowerPC so won’t work unfortunately. The source also looks very windows. 

I’ll give wine a whirl later on.

Fabrizio-Caruso, 2018-02-15:
I get a crash in X1 Millenium emulator with a trivial example:
```
#include <stdio.h>
#include <conio.h>
int main()
{
	printf("Press key\n");
	while(kbhit())
		getch();
	while(!kbhit())
	{ }; 
	getch();
	printf("key pressed\n");
	return 0;
}
```

suborb, 2018-02-15:
Good news is that the xmil.exe works under wine. Bad news is that XBrowser88.exe doesn't - it crashes with a an illegal instruction.

I'm going to have to step out of this one unfortunately.

Fabrizio-Caruso, 2018-02-15:
On the other hand under x1.exe by Takeda I am getting a good (or almost good) behavior:
it does not crash and sort of works but restarts from the beginning. Is this the expected behavior for the little program above?

suborb, 2018-02-15:
Ah, that's a good sign. 

Yes, the exit for X1 compiles is a rst 0 so it will restart.

Fabrizio-Caruso, 2018-02-15:
@all and @suborb, I confirm that getk is not behaving well on my game. No key detected, at all.
If I remove the wait for key press, I get to the game where no key press is ever detected.
The "rest" of the game works. You just see the bad guys kill the player who cannot move.

suborb, 2018-02-15:
That's an odd result set! Here's the source for kbhit:

```
_kbhit:
        ld      hl,(kbhit_key)  ; check if we've got a keypress pending
        ld      a,h
        or      l
        ret     nz              ; we've got something pending
        call    getk            ; read the keyboard
        ld      (kbhit_key),hl
        ret
```

So it calls getk!

For reference, getch is this:

```
getch:
_getch:
        ld      hl,(kbhit_key)
        ld      de,0
        ld      (kbhit_key),de
        ld      a,h
        or      l
        ret     nz              ; consume that keypress
        ; We didn't have anything, lets just read the keyboard
        call    fgetc_cons
        ret
```

And then fgetc_cons is:

```
_fgetc_cons:
        ld hl, _x1_keyboard_io+1
        ld a, (hl)
        and $40
        jp nz, fgetc_cons
        dec hl
        ld a, (hl)
        and a
        jp z, fgetc_cons
        ld (hl), 0

IF STANDARDESCAPECHARS
        cp      13
        jr      nz,not_return
        ld      a,10
.not_return
ENDIF

        ld h,0
        ld l,a
        ret
```

And getk is this:

```
_getk:
        ld hl, _x1_keyboard_io
        ld a, (hl)

IF STANDARDESCAPECHARS
        cp      13
        jr      nz,not_return
        ld      a,10
.not_return
ENDIF

        ld h, 0
        ld l, a
        ret
```

Which makes both paths roughly equivalent!

Fabrizio-Caruso, 2018-02-15:
Ok, my analysis may be wrong but I get this result in my game if I remove the code portion with getch and kbhit, I am left with just getk.
Under x1 by Takeda the game runs normally but no input is detected (I get to the action because no key-press is necessary after removing the wait for key press routine)
Under X1 Millenium, it just crashes.
My assumption is that something is wrong in getk for sure.
getch may be also wrong but it sort of work under one emulator by luck and maybe by the fact that it is used just after kbhit is detected as true.
getch may be wrong as well and it probably is.

zx70, 2018-02-16:
I try by forcing zeroes on getk when a key is not found,

zx70, 2018-02-16:
I just committed a fix, please test it tomorrow after the usual nightly step.
Your sample code now works well on XMillenium but I'm not 100% sure that the keyboard reading will be reactive enough for a game.  In such case I'd try by removing one more line in getk  before giving up  with the quick'n'dirty approach  ;)

Fabrizio-Caruso, 2018-02-17:
The problem persists and I have modified the simple test to better isolate the issue.
If I compile and run the following code, the key presses are either never detected or detection of a single key takes 20-60 seconds:
```
#include <stdio.h>
#include <conio.h>
int main()
{
	int i;
	for(i=0;i<10;++i)
	{
		printf("Press key\n");
		while(kbhit()) getch();
		while(!kbhit()){ }; 
		getch();
		printf("Key pressed: %d\n\n", i);
	}
	return 0;
}

```

zx70, 2018-02-17:
I changed the code a bit already, I'll try your test program soon btw

Fabrizio-Caruso, 2018-02-18:
Still no improvement.
In my game 
1. I can see no movement detected during game play and 
2. the wait for a key-press waits forever.

I have written a simple example that shows how erratic and slow getk is under X1:
```
#include <stdio.h>
int main()
{
	char key;
	while(1)
		if((key = getk())!='\0')
			printf("%c", key);
	return 0;
}
```

zx70, 2018-02-19:
I reverted substantially to the original code and honestly I see no problems running your examples  :(
Please update your z88dk copy tomorrow morning and let me know.

zx70, 2018-02-20:
Note for Dom: I'm trying to embed the boot disk creation in appmake.
In the meanwhile I found X1DiskExplorer which could be a substitute for xbrowser88 ..sadly it is in Japanese!

http://www.geocities.co.jp/ceeezet/index.html

zx70, 2018-02-20:
-create-app option is now available for the Sharp X1   ;)
It creates a simple raw boot disk image (.2D extension) in HuBasic format.  
The conversion to .D88 is possible with XBrowser88 or SamDisk but it is probably not necessary since the emulators support both the formats.

Fabrizio-Caruso, 2018-02-20:
@zx70, nice to have a bootable disk!

I confirm that even reading getk will crash X1 Millenium emulator.
With X1 Millenium I gave tested:
1. my game -> crash at startup once you press a key
2. the above example with just getk -> crash after key press

On the X1 by Takeda, I can run my game if I skip the initial wait for key press.
The game runs but no key presses are detected.

Moreover X1 has an additional issue: sleep "sleeps" for a random time (issue previously present on many other targets)

zx70, 2018-02-20:
I'm sorry but I'm able to run your sample programs with both the emulators and both with "-startup=2" or standard mode, I can still wait for the next nightly build and look for some minimal difference between my local z88dk copy and the current snapshot.

Fabrizio-Caruso, 2018-02-20:
@zx70, sorry for reporting incorrectly the issue.
I was confused by the fact that my game freezes. 
It seems that sleep "sleeps" for a random or very long time (or forever).
If I disable sleep, my game runs fine under X1 by Takeda.

Indeed the keyboard is fine under X1 by Takeda (X1 Millenium crashes).
I do not use -startup=2.

Fabrizio-Caruso, 2018-02-20:
I may open a different issue about the sleep routine under X1.

closed 2018-02-20 by Fabrizio-Caruso

zx70, 2018-02-20:
perfect, I'm closing this one, which already brought few interesting results   :)

Fabrizio-Caruso, 2018-02-20:
Remark: 
1. Under X1 Millenium: my game crashes after a key press
2. Under X1 by Takeda: sleep has to be avoided but the rest of the game works fine

suborb, 2018-02-20:
I'll rework sleep to use the newlib asm_delay_ms() function shortly - at the moment it relies on a ticks counter which may not be available on all platforms.

zx70, 2018-02-20:
I just found a possible bug subcpu communications handshake which might compromise the date/clock commands on the real hardware or well built emulators.   I'm fixing now, but the delay granularity for X1 is of a whole second (no ms, at least in the current implementation).

suborb, 2018-02-20:
@Fabrizio-Caruso Rather than change sleep, I've added msleep() (from stdlib.h) that should be pretty accurate.

If it's not accurate then that's because I've got the wrong clock speed for a target! I'll wrap it up into sleep() at some point later. 

@zx70 there's some "funky" stuff for c128 and zx80 for sleep(). For the c128 I think we can just switch over to the T-state wait that is msleep(), but I'm not certain what's going on with the zx80

Fabrizio-Caruso, 2018-02-20:
@suborb, sleep seems totally wrong, if it cannot be changed it should not be made available. It behaves randomly: it may not pause at all, pause a random time or pause forever.
It looks like an old issue that was affecting most targets.

zx70, 2018-02-21:
I'm fixing the subcpu communications a little more, but it looks ok already.   Please test with sleep(1) to have integer values.

suborb, 2018-02-21:
@Fabrizio-Caruso I've gutted the sleep/csleep/delay code. For all(*) platforms it's sleep/csleep is now expressed in terms of msleep() and hence an exact T state delay based on the z80 clock speed. I've removed delay since it's the same as msleep.

(*) ZX80 retains the old code, z88 uses the custom code it had before since blocking the OS out will burn through batteries.