z88dk/z88dk #611: For discussion: z80asm send errors to stderr instead of file.err

pauloscustodio, 2018-02-20:
Currently z80asm creates a file.err when there are compilation errors.

Proposal: remove the generation of .err files, send errors only to stderr 

Any comments?

suborb, 2018-02-20:
zcc does some work to look up the source file for the error and show the line where the problem is. I think this includes a translation back to the original C file.

Without the .err files this isn't going to be possible.

However, one could argue that if z80asm were to report the line and print the offending assembler, then the user can look up the C file and line number themselves.

pauloscustodio, 2018-02-20:
If zcc emits C_LINE statements, then z80asm will report errors in the original C file lines.

suborb, 2018-02-20:
Yes, so at the moment, we parse the .err files and print the offending C source line.

Given that we use system() we can't capture stderr so can't do the same thing. I don't think we can even do it portably with popen()

pauloscustodio, 2018-02-20:
Clear. Idea abandoned.

suborb, 2018-02-20:
I have to admit that I'm not keen on the .err file, it's a little bit of QDOS hangover really.

The logic for printing the line could moved into z80asm though - it's currently in zcc.c::ShowErrors

zx70, 2018-04-15:
is this issue connected to #688  ?

pauloscustodio, 2018-04-15:
I think so although I did not yet investigate further. 

⁣Enviado por TypeApp ​

pauloscustodio, 2022-01-01:
Starting work on this: showing offending assembly line after error, and getting rid of Perl generating source files.

While at ir: is the error message format carved in stone, or could we change to the more terse and more standard?
```
file:line: error: xxxx
```

suborb, 2022-01-01:
A more standard format would be welcome - would be great if vscode etc would just be able to parse them.

commit 21349cc4 (2022-01-05): Fix #611 remove .err files
Remove .err files, errors are sent to stderr only
Show assembly line that caused the error in pass 1; show expression that caused the error in pass 2 or during linking
The function ShowErrors in zcc.c can be removed - it will find no .err files to parse
Remove dependency on z80pack z80asm assembler - was being used for testing
Fix all the tests
closed 2022-01-05 by pauloscustodio