z88dk/z88dk #614: [sms] Broken target? Black screen? [tested with Z88DK example chicken.c]

Fabrizio-Caruso, 2018-02-22:
I am trying to use Z88DK to compile for the sms (as well as all other targets).

I am getting a black screen at startup and nothing else even with the examples found in Z88DK.

I compile with:
```
sms_chicken:
	$(Z88DK_PATH)$(MYZ88DK) +sms \
	-vn -lndos \
	-o $(BUILD_PATH)/sms_chicken.sms \
	$(SOURCE_PATH)/../experiments/chicken/chicken.c \
	$(SOURCE_PATH)/../experiments/chicken/chicken_graphics.asm	
```
I get sms_chicken.sms, which I try to load into the Fusion emulator (Fusion364-1995) by
File -> Load MasterSystemROM and I get a black screen.

Is the target broken? Am I doing something wrong?

suborb, 2018-02-22:
I've just tried this out in Emulicious and chicken works.

For SMS emulation, Emulicious is apparently the emulator to use given its accuracy.

Fabrizio-Caruso, 2018-02-22:
@suborb, I get a black screen with Emulicious as well.
I guess I am not compiling with the right options.
What are the correct options to get a usable .sms file?

suborb, 2018-02-22:
I just use the makefile in the examples! 

However I can see that you're missing the -create-app option. So try this:

```
sms_chicken:
	$(Z88DK_PATH)$(MYZ88DK) +sms \
	-vn  \
	-o $(BUILD_PATH)/sms_chicken.bin -create-app \
	$(SOURCE_PATH)/../experiments/chicken/chicken.c \
	$(SOURCE_PATH)/../experiments/chicken/chicken_graphics.asm	
```

That should create a .sms file that can be loaded

Fabrizio-Caruso, 2018-02-22:
@suborb, sorry and thanks! 

I close this non-issue.

I have a question though: does Z88DK with +sms support the Game Gear and SG-1000, as well?
It should be possible because devKit does support the Game Gear and SG-1000 within the same toolkit and devKit was sort of integrated into Z88DK.

closed 2018-02-22 by Fabrizio-Caruso

suborb, 2018-02-22:
I don't think AA pulled in the game gear bits of devKitSMS into newlib. There's certainly nothing explicit in classic to support those platforms either.

aralbrec, 2018-02-22:
devkitSMS is integrated into newlib; there are examples [here](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms).  Astroforce is a good example and the same author has done another game recently called [Silver Valley](https://www.youtube.com/watch?v=z7QBm97X0go) using z88dk.

aralbrec, 2018-02-22:
Pressed send by mistake there....

The classic library has another tms api that I think was originally written by haroldoop years ago.

It's only the sms portion of devkitSMS that has been integrated into newlib.  I wasn't sure if the game gear should get a separate target so I put that off for later.

Fabrizio-Caruso, 2018-02-23:
@aralbrec, Thanks!
So classlib uses different APIs than newlib, which instead uses the sms portion of devkitSMS and its APIs?

How do I compile sms with newlib?
This very simple thing is still a mystery for me.
I have managed to compile with newlib + zsdcc by using different options depending on the target.
I understand that some targets need to preserve different registers.
Is there a general rule?

aralbrec, 2018-02-23:
I'm not sure why your message wasn't reflected here so I'll just quote it:

> So classlib uses different APIs than newlib, which instead uses the sms portion of devkitSMS and its APIs?

Yes that's right plus a few other things like text terminals and functions to copy to/from vdp memory directly.

> How do I compile sms with newlib?
> This very simple thing is still a mystery for me.

For any target, if "-clib=new", "-clib=sdcc_ix" or "-clib=sdcc_iy" appears in the compile line then newlib will be used.  Of course classic supports many more targets than newlib.

For the sms + newlib target specifically, the examples here give typical compile lines:
https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms

Some of these programs occupy many memory banks so sometimes portions of the program are compiled separately to individual memory banks before being linked together to make the final binary.

[Astroforce](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms/AstroForce) is an example.  You can see the list of zcc invokes used to compile that program.

If you're program fits in 32k rom then you don't have to do any banking.  Like [speech bubbles](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms/speechbubbles) where there is one zcc invoke.

The startup values control what sort of crt is used.  Some are devkitSMS crts and some do not have devkistSMS related things in the crt.  This is described in [helloworld](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms/helloworld).

The sms starts like a blank slate.  There's nothing there, just your program.  So you'll see the example `helloworld` load a font into vdp ram in order to be able to print.  The text terminals are newlib terminals which operate pretty much the same across all newlib targets.

> I have managed to compile with newlib + zsdcc by using different options depending on the target.
> I understand that some targets need to preserve different registers.
> Is there a general rule?

You always want to use a "-clib=sdcc_iy" compile in newlib or "-compiler=sdcc --reserve-regs-iy" in classic because this gives ix to sdcc for its frame pointer, it gives iy to the library and it changes the code generation pattern so that it's more amenable to the -SO3 peephole rules.  You get better code out.

Some targets have restrictions associated with them.  Like the zx80, zx81 which use certain registers when generating the video display.  On these machines, the z80 cpu is actually generating the display so you have to leave certain registers alone.  But there are no such targets in the newlib.  With classic lib, you may have to be careful when compiling with zx80/zx81 using zsdcc but there has been success recently doing it.  The classic library has some accommodations for the zx80/zx81 whereas the newlib does not.