z88dk/z88dk #617: zsdcc: please , provides a non-zipped sdcc-z88dk.patch ...

bignaux, 2018-02-23:
... it would improve packaging of it for nix. Or at least, not a flatted one.

aralbrec, 2018-02-24:
The patch is now maintained in z88dk/src/zsdcc

closed 2018-02-24 by aralbrec

aralbrec, 2018-02-24:
> the patch is same as https://github.com/z88dk/zsdcc branch zsdcc ?

Yes.  With the patch the build of zsdcc involves an svn checkout of sdcc's source at revision 9958 (we haven't moved passed this yet because we had code generation errors after this), apply the patch and build.  There will be errors while sdcc builds because the changes are not compatible with some of sdcc's cpu targets but all that's important is that the "sdcc" and "sdcpp" binaries are built; those are copied and renamed to "zsdcc" and "zsdcpp".  I haven't done it myself but you should be able to configure the sdcc build to only do the z80 family (z80/z180/rabbit/gbz80) so that the build process is sped up and finishes without errors.

https://github.com/z88dk/zsdcc is a reflection of my local source code tree that compiles zsdcc using vs2015.  This has to be cleaned up so that it can compile using the standard .configure as well as vs2015 and the plan is to clean up the changes we've made so that many of them can be submitted upstream.  Eventually we'll be including this as a submodule in z88dk/z88dk itself.

bignaux, 2018-02-24:
for the moment i did that https://github.com/NixOS/nixpkgs/pull/35463 , since i've some difficulty to get official svn of sdcc, and then it simplify my nix derivation. i did  z88dk unstable earlier (merge in nixpkgs) so we will have a complete solution for linux user ( read https://nixos.org/nix/ , all linux users can use it).

aralbrec, 2018-02-24:
> i've some difficulty to get official svn of sdcc

It sounds like sf migrated to new servers and there may be some temporary hiccups.

> The snapshots page is back only sometimes, usually some of the snapshots
> are missing now. The svn currently seems to work via svn+ssh, but at up
> to yesterday it wouldn't work reliably for me using https.

commit cbc9a6d8 (2018-02-24): Fix #617: supply zsdcc patch in unzipped form in z88dk/src/zsdcc