z88dk/z88dk #618: +sms (classic) 3d city is broken

suborb, 2018-02-24:
The music appears a little odd, no aliens appear and the cursor doesn't move.

suborb, 2018-02-24:
Commenting out the music makes it work once more.

suborb, 2018-02-24:
Music uses $c100 to $c300 as fixed addresses. There's a bit of hack to reserve 1k for this, but this isn't lining up correctly:

Section 'BSS' size: 1 bytes ($C000 to $C000)
Section 'bss_fp' size: 0 bytes ($C001 to $C000)
Section 'bss_error' size: 2 bytes ($C001 to $C002)
Section 'bss_crt' size: 155 bytes ($C003 to $C09D)
Section 'bss_fardata' size: 0 bytes ($C09E to $C09D)
Section 'bss_compiler' size: 117 bytes ($C09E to $C112)
Section 'bss_clib' size: 0 bytes ($C113 to $C112)
Section 'bss_string' size: 0 bytes ($C113 to $C112)
Section 'bss_alloc_balloc' size: 0 bytes ($C113 to $C112)
Section 'bss_user' size: 1024 bytes ($C113 to $C512)

I'll need to strip down bss_crt a little by reducing the size of the stdio block

closed 2018-02-24 by suborb

aralbrec, 2018-02-24:
You can just change the bss org to be past the reserved area with a pragma

suborb, 2018-02-24:
Yeah, that would've been easier :) #pragma-define:CRT_ORG_BSS=0xc300 would have done the trick

commit bc02b623 (2018-02-24): Fix #618: Try and make bss_user start before $c100 so that the
music doesn't trash variables