z88dk/z88dk #621: cp/m option '-startup=2' for reloc failing

zx70, 2018-02-28:
I'm using this option to test the graphics code on the keypro'83, it worked already on the tiki 100 but recently it had to be changed due to the segments introduction.
 My past fix was less than precise,  now it works better but I think that the bss segmnt and maybe something else are still pointing to the lower memory halve.  Can appmake be part of the problem?

suborb, 2018-02-28:
There's no appmake for CP/M - the output of the assembler is the actual binary.

zx70, 2018-03-01:
I think we have appmake.. it is just renaming the file and adding the extra sections ?

aralbrec, 2018-03-01:
The default for cpm has appmake rename the output file.
cpm is exclusively a ram model compile so the linker puts together the binary.

zx70, 2018-03-01:
checking the map file confirms what you said .

zx70, 2018-03-02:
Solved, it was a bug in the display addressing code for lorez gfx.    Another bug related to 'startup=2' existed in the relocation code but I solved it before opening this issue.

closed 2018-03-02 by zx70