z88dk/z88dk #630: zpragma is puting out two defines for pragma output into zcc_opt.def

aralbrec, 2018-03-06:
I just noticed this from compiles using a pragma file and -pragma-include on zcc's compile line:

```
IF !DEFINED_CRT_ORG_CODE 
	defc	DEFINED_CRT_ORG_CODE  = 1
	defc CRT_ORG_CODE  = 0x6000
	IFNDEF CRT_ORG_CODE 
	ENDIF
ENDIF

IF !DEFINED_CRT_ORG_CODE 
	defc	DEFINED_CRT_ORG_CODE  = 1
	defc CRT_ORG_CODE  = 0x34304
	IFNDEF CRT_ORG_CODE 
	ENDIF
ENDIF

IF !DEFINED_CRT_ORG_BSS  
	defc	DEFINED_CRT_ORG_BSS   = 1
	defc CRT_ORG_BSS   = 0xffffffff
	IFNDEF CRT_ORG_BSS  
	ENDIF
ENDIF

IF !DEFINED_CRT_ORG_BSS  
	defc	DEFINED_CRT_ORG_BSS   = 1
	defc CRT_ORG_BSS   = 0x34304
	IFNDEF CRT_ORG_BSS  
	ENDIF
ENDIF
```

Every `#pragma output` in the pragma file gets a second garbage define with the same value 0x34304.  `#pragma redirect` and `#pragma printf` in the same file don't get this.

It doesn't affect the compile because the first (correct) define is used.

suborb, 2018-03-06:
The cause is some comically bad handling of stackptr

closed 2018-03-07 by suborb

zx70, 2018-03-08:
<blush>... thank you, Dom !