z88dk/z88dk #640: [PX8] Error with no specified line

Fabrizio-Caruso, 2018-03-13:
I am trying to compile for the PX8 target (+cpm -subtype=px32k) and I am getting 
an error with no specified line. How can I get a more informative error message?

```
/cygdrive/c/z88dk/bin/zcc.exe +cpm -subtype=px32k \
-D__PX8__ \
-DTINY_GAME \
-create-app -obuild/TINY_px8.bin \
src/display_macros.c \
src/enemy.c \
src/level.c src/character.c src/text.c  \
src/strategy.c src/input_macros.c \
src/main.c

PROCESSING src/display_macros.c
ucpp -iquote"." -D__Z88DK -D__CPM -DCPM -D__CPM__ -DZ80 -D__Z80 -D__PX8__ -DTINY_GAME  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"c:/z88dk//include"    "src/display_macros.c" "C:\cygwin64\tmp\zcc117A2.i2"
zpragma -sccz80 < "C:\cygwin64\tmp\zcc117A2.i2" > "C:\cygwin64\tmp\zcc117A2.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "C:\cygwin64\tmp\zcc117A2.i"
make: *** [Makefile:966: px8_tiny] Error 1
```

suborb, 2018-03-13:
That usually means that sccz80 has crashed. 

The usual cause of that is an unknown variable when sccz80 tries to carry on. I’ve got a few changes in my workspace area in this area so I’ll try to push one of them this evening.

suborb, 2018-03-13:
Yup, that was it:

sccz80:"src/display_macros.c" L:163 Error:Unknown symbol: COLOR_CYAN
sccz80:"src/display_macros.c" L:164 Error:Unknown symbol: COLOR_RED
sccz80:"src/display_macros.c" L:168 Error:Unknown symbol: COLOR_YELLOW
sccz80:"src/display_macros.c" L:169 Error:Unknown symbol: COLOR_GREEN
sccz80:"src/display_macros.c" L:170 Error:Unknown symbol: COLOR_BLUE
sccz80:"src/display_macros.c" L:228 Error:Unknown symbol: COLOR_WHITE

On Linux/Mac you get the first error message so it's not quite so noticeable, but on windows it looks like it gets swallowed.

Regardless, it should now be good.

suborb, 2018-03-16:
@Fabrizio-Caruso I see you've got compilation working for PX-8 now, so I'll close this issue.

closed 2018-03-16 by suborb

Fabrizio-Caruso, 2018-03-16:
@suborb THANKS a lot. Yes, indeed I have even managed to get to compile my game for this platform!
There may be some issues with sleep and kbhit/getk with this target. I need to verify this before I open a new issue.