z88dk/z88dk #643: question sccz80 yaz180 - missing linkage for library functions

feilipu, 2018-03-14:
As part of getting set up for using sccz80 far calls as per #596 and #641, I've been doing some preparation.

I compiled the ChaN [FatFS code](http://elm-chan.org/fsw/ff/00index_e.html) using sccz80. This year, with [almost no issues](https://github.com/feilipu/z88dk-libraries/commit/a5b07a8082c858e2f27ad1f119183bc7fb006936) the code worked out of the box. This is a huge improvement over the "old" sccz80 from last year. :+1: :100: 

After [adding these time and ff libraries](https://github.com/feilipu/z88dk-libraries) to my z88dk install using z88dk-lib, I tried to compile [yaz180 yabios](https://github.com/feilipu/yaz180/tree/master/yabios) using a similar command to what I would normally use for sdcc. 

```bash
zcc +yaz180 -v -subtype=yabios -SO3 --list -m --c-code-in-asm -clib=new -llib/yaz180/ff -llib/yaz180/time  main.c -gpf yabios.rex -o yabios -create-app
```
That step results in a large number of errors for things that I would have expected would be defined.
Items from both the 3rd party libraries, and the z88dk internal libraries.

[sccz80_yabios_compile_errors.txt](https://github.com/z88dk/z88dk/files/1811169/sccz80_yabios_compile_errors.txt)

I guess I'm missing a step... any hints welcome, please.

suborb, 2018-03-14:
I suspect this may be a result of compiling the libraries with sdcc and the routines not being exported for sccz80 as a result.

If a library routine is marked with `__LIB__` then sccz80 expects to find the library routine without a leading underscore, whereas sdcc will ignore `__LIB__` and export the symbol with a leading underscore.

After the changes with z80asm over the years, this distinction between project and library symbols is not required, however (in some issue I can't find at the moment) we opted to keep the `__LIB__` specifier for sccz80 - it helps to distinguish between `__smallc` and `__stdc` calling convention entry points.

If you keep the libraries as compiled with sdcc then you'll need to two things: remove the `__LIB__` decorator, and then add a `__stdc` decorator at the end. That will ensure that sccz80 calls the underscore function name and will push the parameters in the correct order.

suborb, 2018-03-14:
I took a look at the checked in ff.lib and it's not quite as I described above. The library is compiled with sccz80, but the private header used during the library compilation doesn't have the `__LIB__`.

However, I'm guessing that the public header used for the yabios compilation does have `__LIB__`. 

So in that case, the resolution is to drop the `__LIB__` from the public headers

aralbrec, 2018-03-14:
Yes the LIB thing is the issue.  The headers are being defined for the PROTO directory in the new lib from which headers will be generated automatically for zsdcc and sccz80.  For sccz80 those headers will include LIB directives for all the functions.  As mentioned this means sccz80 will look for function names without a leading underscore when in fact the functions are compiled with a leading underscore.

What we need to do is add an m4 define like "__Z88DK_NO_LIB" that can be placed in the PROTO headers so that the LIB decorator can be turned on and off within a file when the headers are generated.  I'll take a look at the m4 script tonight or you can have a look at [z88dk/include/_DEVELOPMENT/__link__.m4](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/__link__.m4).

**asm**

Besides that there are other issues for your asm functions.  sccz80 wants to do L->R parameter passing by default which is the opposite order your asm functions would expect if written for zsdcc.  Also, char parameters are passed as two bytes by sccz80 compared to the one byte by zsdcc.  That is coped with in the library by defining two C entry points :- one for zsdcc and one for sccz80 - that jump into the asm implementation.  Part of the reason for retaining the LIB decoration is keeping the future possibility of mixing zsdcc and sccz80 object code and that LIB decoration would make sure sccz80 and zsdcc still have different entry points from C.

You can change the parameter order to R->L in sccz80 on a per function basis with suitable decorator on the function prototype (again maybe another define for m4) but you still have to cope with the 1-byte char versus 2-byte char issue.  In the classic library many functions were changed so that params were char instead of int for this reason.

One last thing, if the return value is char, sccz80 expects two bytes returned in HL with H=0.

suborb, 2018-03-14:
Actually, `__z88dk_sdccdecl` should take care of both the R->L and single byte char handling for both calling and the return value from functions.

Classic can't use that decorator since all library functions are defined as `__smallc`.

I can feel some docs coming up on calling conventions and the associated decorators  - I'll tie it in with #641

feilipu, 2018-03-15:
Ok. I will leave it till the docs arising from #641 are done.

I wasn't too concerned by the L->R or R->L issues, as I knew that these may (would) need to be fixed in my yabios assembly function preambles. My surprise was around the multitude of linking errors, so it is good to hear that it is not just me. 😄

suborb, 2018-03-15:
I've started some words here: https://github.com/z88dk/z88dk/wiki/CallingConventions

feilipu, 2018-03-15:
`__z88dk_shortcall(RR, VV)` VV >256 should be `defw`.

aralbrec, 2018-03-16:
I added an m4 macro to eliminate the `__LIB__` decorator in sccz80 prototype functions.  You can define it to disable the decorator and then undefine it to re-enable.

So a header prototype could look like this:

```
define(`m4_SCCZ80_NOLIB')
..function prototypes as usual with no LIB for sccz80
undefine(`m4_SCCZ80_NOLIB`)
..more function prototypes with LIB for sccz80
```

The define/undefine could also be pushdef/popdef if stack behaviour is needed.

feilipu, 2018-03-19:
Where does this leave the process for linking 3rd party libraries compiled with `sccz80` and installed with `z88dk-lib`?

Does either the library, or the application compile need to do something with `m4_SCCZ80_NOLIB`, or is it now an automatic process?

Thanks.

aralbrec, 2018-03-20:
Just put the define at the top of the header file:

```
include(__link__.m4)
define(`m4_SCCZ80_NOLIB', 1)
```

This will cause the sccz80 prototypes to be made without the `__LIB__` decoration.

feilipu, 2018-03-26:
Fixed, with thanks.

I still have to fix linkage for my asm functions being called by the library concerned.
But that's my issue, and another days' work.

closed 2018-03-26 by feilipu