z88dk/z88dk #649: z80asm: symbol declared public and not defined should raise an error

pauloscustodio, 2018-03-17:
A symbol declared public and not defined should not be allowed.

Include files should use global - this either declares the symbol as public if defined, or as extern otherwise.

pauloscustodio, 2018-03-17:
see also #650

pauloscustodio, 2018-03-17:
Fails the library build: 
```
Error at file '../_DEVELOPMENT/z80/z80/asm_z80_pop_ei.asm' line 22: symbol 'asm0_z80_pop_ei' not defined
Error at file '../_DEVELOPMENT/z80/z80/asm_z80_pop_ei.asm' line 23: symbol 'asm_z80_pop_ei_jp' not defined
Error at file '../_DEVELOPMENT/z80/z80/asm_z80_pop_ei.asm' line 26: symbol 'asm0_cpu_pop_ei' not defined
Error at file '../_DEVELOPMENT/z80/z80/asm_z80_pop_ei.asm' line 27: symbol 'asm_cpu_pop_ei_jp' not defined
```

symbol is declared public but if-def does not include it's definition.

Should we fix the library, or revert to without this check?

suborb, 2018-03-18:
Needs to be a noop. It isn’t actually called on the rabbit - ipset and ipres are used instead.

aralbrec, 2018-03-19:
There may be library functions that call these routines to save and restore interrupt state so I filled in the rest of the rabbit equivalents.  In the end most of these lightweight subroutine calls should be inlined by the peepholers #595.

pauloscustodio, 2018-03-21:
I've been fixing the library to pass the compilation, although I will need someone, maybe @aralbrec to check if my changes are correct. I'll request a peer-review before merge.

Anyway there is one I don't how to solve: 
```
z80asm  -xlib/sccz80/zx @target/zx/library/zx_sccz80.lst
Error at file 'arch/zx/bifrost2/z80/BIFROST2_INSTALL.asm' line 1042: symbol 'asm_BIFROST2_stop' not defined
```

The file BIFROST2_INSTALL.asm says that it is  Automatically Generated at Library Build Time. Where is the source?

aralbrec, 2018-03-22:
> The file BIFROST2_INSTALL.asm says that it is Automatically Generated at Library Build Time. Where is the source?

It's made from the m4 file in the same directory:  `arch/zx/bifrost2/z80/BIFROST2_INSTALL.asm.m4`

commit 97bc7b95 (2018-03-25): Fix #649: raise error if symbol declared public and not defined
closed 2018-03-25 by pauloscustodio