z88dk/z88dk #650: z80asm: symbol declared public and then extern should raise an error

pauloscustodio, 2018-03-17:
A symbol is either public or extern; if both declarations appear for the same symbol, an error should be raised.

Include files should use global - this either declares the symbol as public if defined, or as extern otherwise.

see also #649

pauloscustodio, 2018-03-17:
Bad idea, breaks old code.

Leave as is, extern after public is just ignored.

closed 2018-03-17 by pauloscustodio

suborb, 2018-03-17:
Any pointers as to which bit of code breaks?

pauloscustodio, 2018-03-17:
Several of my test cases failed, and the code has an explicit comment "ignore extern if symbol already public". 

Do you think we should have this check? If yes, I can try to check why the test cases fail.

aralbrec, 2018-03-18:
Quite often prototypes are declared extern in C and they get translated to extern statements in asm.  Then an implementation file includes its own header so that the prototype generates an extern and the actual definition generates a public in the same file.

Using global for the prototypes misses a class of errors where the .c file is intended to provide a local definition of a function but, if misnamed or not present, a global definition is grabbed instead because of the global directive used with the prototype.