z88dk/z88dk #652: z80asm: implement macros

pauloscustodio, 2018-03-18:
[ ] #define name text ; name is replaced by text in all further text
[ ] #undef ; undefine a macro
[ ] #define name(arg1, arg2, ...) text ; name aaa, bbb -> aaa and bbb are collected and inserted as arguments and macro expanded
[ ] #defcont xxxx ; expand previous #define macro with a new line of text
[ ] macro name arg1, arg2 / endm ; define multi-line macro
[ ] name macro arg1, arg2 / endm ; alternative syntax
[ ] name macro / endm ; alternative syntax, arguments can be used as %1, %2, ...

suborb, 2021-10-29:
Not good news I'm afraid:

```
make
x86_64-apple-darwin15-c++ -g -Wall -Wextra -Werror -pedantic-errors -std=gnu++17 -MMD    -c -o z80asm-pp.o z80asm-pp.cpp
error: invalid value 'gnu++17' in '-std=gnu++17'
<builtin>: recipe for target 'z80asm-pp.o' failed
make: *** [z80asm-pp.o] Error 1
build@build2:~/src/z88dk/src/z80asm-pp$ x86_64-apple-darwin15-c++ -v
clang version 3.8.1-12ubuntu1 (tags/RELEASE_381/final)
Target: x86_64-apple-darwin15
Thread model: posix
```

Native (and mingw):

```
g++ -g -Wall -Wextra -Werror -pedantic-errors -std=gnu++17 -MMD    -c -o utils.o utils.cpp
In file included from utils.cpp:7:0:
utils.h:13:22: fatal error: filesystem: No such file or directory
 #include <filesystem>
                      ^
compilation terminated.
```

```
g++ -v

....

gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12)
```

pauloscustodio, 2021-10-30:
Any chance to upgrade to gcc 7.1 or greater?

pauloscustodio, 2021-10-30:
@suborb, I have replaced std::filesystem from C++17 by a submodule ghc::filesystem that should compile in C++14.
Can you please test the nightly build with branch issue652?

Thanks.

suborb, 2021-10-30:
It looks like I'm going to have to completely rebuild the container image used for builds:

Native (and presumably mingw) build:

```
re2c -W --tags --no-debug-info --no-generation-date  -o cmdline.cpp cmdline.re
re2c: error: bad long option: --tags
Makefile:58: recipe for target 'cmdline.cpp' failed
make: *** [cmdline.cpp] Error 1
```

macos build doesn't even get that far:

```
x86_64-apple-darwin15-c++ -g -Wall -Wextra -Werror -pedantic-errors -std=gnu++14 -MMD -I../../ext/filesystem/include    -c -o utils.o utils.cpp
In file included from utils.cpp:7:
In file included from ./utils.h:13:
../../ext/filesystem/include/ghc/filesystem.hpp:194:10: fatal error: 'chrono'
      file not found
#include <chrono>
         ^
1 error generated.
<builtin>: recipe for target 'utils.o' failed
make: *** [utils.o] Error 1
build@build2:~/src/z88dk/src/z80asm-pp$ make
x86_64-apple-darwin15-c++ -g -Wall -Wextra -Werror -pedantic-errors -std=gnu++14 -MMD -I../../ext/filesystem/include    -c -o symtab.o symtab.cpp
In file included from symtab.cpp:8:
./symtab.h:11:10: fatal error: 'unordered_map' file not found
#include <unordered_map>
```

suborb, 2021-10-31:
I'm in the process of rebuilding the container - `z80asm-pp` is now linking for macOS and win32/win64.

I still need to:

- [ ] Install additional dependencies needed for other tools
- [ ] Invoke the whole build manually to ensure it builds
- [ ] Verify what MacOS version we're now targeting
- [ ] Check the the right dlls are shipped in the windows kit
- [ ] Possibly rebuild the win32 sdcc binaries to use the same dlls
- [ ] Set up build schedule on `build3` (new container)
- [ ] Decommission `build2` (old container)

pauloscustodio, 2021-10-31:
Is it ok to revert to using C++17?

suborb, 2021-10-31:
Looks like we're good - I just rolled back a change and z80asm-pp linked.

suborb, 2021-11-28:
I'm sorry, the real world has been kicking me hard lately so I've not looked into this further. Consider this post bumping it up my list.

pauloscustodio, 2021-11-29:
No problem. The solution I had envisaged with a separate preprocessor hit a blocking point: the IF expressions today may refer to labels defined before the IF statement. These constructs failed in the new solution because the preprocessor was completely separate from the code generation.

The new strategy I am implementing (albeit slowly) is to integrate the preprocessor inside z80asm. The new modules are in C++17, integrated with the old C source. So, as long as we can compile with gcc/g++ 7.1 or greater, this solution will work.

pauloscustodio, 2021-12-28:
z80asm is finally a macro assembler!

Implemented MACRO/ENDM, REPT/ENDR, REPTC/ENDR, REPTI/ENDR, EXITM, LOCAL, DEFL, #define, #undef

Documentation at https://github.com/z88dk/z88dk/wiki/Tool---z80asm---preprocessor

closed 2021-12-28 by pauloscustodio