z88dk/z88dk #656: [M5][question][issue?] Probable stack issue (__register_sp' not defined)

Fabrizio-Caruso, 2018-03-20:
When compiling my game for the m5 target with
```
m5_tiny:
	$(Z88DK_PATH)$(MYZ88DK) +m5 \
	-lm -create-app -Cz--audio \
	-D__M5__ \
	-DTINY_GAME -DLESS_TEXT -DNO_SLEEP \
	-o$(BUILD_PATH)/TINY_m5.bin \
...
```
I get the following error:
```
...
GENERATING OUTPUT

PROCESSING CRT
z80asm -I"c:/z88dk/\lib\target\m5\def"  -s  --cpu=z80 -I"c:/z88dk//lib" -I"c:/z88dk/\lib\target\m5\classic"  "C:\cygwin64\tmp\zccC61910.asm"
Errors in source file c:/z88dk/\lib\target\m5\classic\m5_crt0.asm:
Error at file 'c:/z88dk/\lib\target\m5\classic\m5_crt0.asm' line 50: unclosed quoted string
                   ^ ----         INCLUDE "crt/classic/crt_rules.inc
Error at file 'c:/z88dk//lib/crt/classic/crt_init_sp.asm' line 9: symbol '__register_sp' not defined
                   ^ ----       ld sp,__register_sp      ; stack is at fixed address
make: *** [Makefile:991: m5_tiny] Error 1
```

suborb, 2018-03-20:
Sorry, I thought there was an example that built for the m5, it appears not - there's a missing quote in m5_crt0.asm - I'll fix it tonight.

Fabrizio-Caruso, 2018-03-20:
@suborb thanks a lot!
As you may have understood I am writing a universal 8-bit abstraction layer (and a game the uses it) for all targets of all 8-bit cross-compilers.
I could make a simplified version of my project as a non-regression for Z88DK (and other tool kits) at least for test the compilation. I cover more targets than current examples that are provided with Z88DK.
If this is accepted, I would be glad to contribute.

suborb, 2018-03-20:
@Fabrizio-Caruso that would be great - I've been tempted pull in cross-chase as a dependency and build targets as a sanity check. However a smaller (less volatile/fewer changes) project would work much better.

closed 2018-03-20 by suborb

Fabrizio-Caruso, 2018-03-21:
@suborb that is why I wrote that I could make "simplified" version. I would fork the current ever-growing cross-toolkit project and make a smaller Z88DK-only version that I would not have to update except for bug-fixes and support for more Z88DK-targets. I would rarely back-port new game or framework features that I would develop in the full cross-toolkit project.

commit cbb744be (2018-03-20): Fix #656: Missing quote preventing m5 from compiling