z88dk/z88dk #657: [trs80] How to create and load a correct .cmd file?

Fabrizio-Caruso, 2018-03-20:
If I compile with
```
trs80_tiny:
	$(Z88DK_PATH)$(MYZ88DK) +trs80 -lndos -subtype=disk \
	-lm -create-app \
	-D__TRS80__ \
	-DTINY_GAME -DLESS_TEXT -DNO_SLEEP \
	-o$(BUILD_PATH)/TINY_trs80.bin \
...
	rm $(BUILD_PATH)/TINY_trs80.bin
```

I get a probably corrupted or incomplete .cmd file, which cannot be loaded.

With TRS32 1.28 (2014 by Matthew K Reed) if I select 
File->Run/CMD I get an error message that says 
```
WHAT?
```

Is the target working or am I making some mistake in the compilation or load procedure?

suborb, 2018-03-20:
There's a post on the forum which has an example that runs on the trs80: https://www.z88dk.org/forum/viewtopic.php?id=10250

In essence:

    zcc  +trs80 -lndos -lm -create-app -o plottest_trs80.bin -subtype=disk plottest.c

Note: use a TRS-80 Model III, type "system" to exit Basic and run the .cmd file. Clearing the screen does not work, everything else does.

Fabrizio-Caruso, 2018-03-20:
@suborb, I am compilingin the very same way and my code should run on any target (it is using plain ANSI C).
I just need to know how to load it.
The binary I produce does not run on
 TRS32 1.28 (2014 by Matthew K Reed) 
Is this a recommended emulator for Z88DK executables?

suborb, 2018-03-20:
I can't get that emulator to work properly under wine, but I had success with: http://48k.ca/trs80gp.html

You may want this change: f44fbf4c76094f058d78a04d5b3bf1f60b238d2b

[image]

Fabrizio-Caruso, 2018-03-24:
Under trs32 (which seems the most common emulator), I have so far only managed to get Model I to start (I cannot find good roms or good names for the roms for other models).
If I create a .cas I can load the game with CLOADM. Is it the right command?
Once the game has finished loading, I type RUN and I get random characters on the screen.

Fabrizio-Caruso, 2018-03-24:
trs80gp works indeed with the .cas file produced by Z88DK!

I now have a different problem.
How do I display a character at a given position?
It would be enough to know which escape codes I can use for this target.

The following esc code (to move cursor) has not worked
printf("%c%c%c%c",27,'=',y+32,x+32);

suborb, 2018-03-25:
It looks like there’s no native way to move the cursor to the required position apart from moving one character at a time in ea j direction. 

However it looks like the video memory is accessible so it should.

suborb, 2018-03-25:
...be possibly to write a screen driver that bypasses the bios and writes directly to that memory.

Fabrizio-Caruso, 2018-03-25:
@suborb, Indeed, screen memory is mapped into RAM at $3c00-$3cff. So it should be easy for me to do character graphics by justing "poking" at the right locations. I need to try it fist.

suborb, 2018-03-25:
Okay, if it works for you then I might do a generic "text display" driver for all the character mapped targets so that a standard set of control codes is accepted across the machines (probably based on the Spectrum codes if I'm honest).

suborb, 2018-03-25:
I'm going to close this one now you've got something loaded, the console driver is in #666

closed 2018-03-25 by suborb

Fabrizio-Caruso, 2018-03-25:
Are you going to change the escape codes for machines that already have them?
It may be a good idea but it would break code and it would use more memory in those targets that has a specif set of escape codes in their bios/basic/rom.
Which Spectrum escape codes do you mean? The Specteum has two different ones (class vs newlib).

Fabrizio-Caruso, 2018-03-25:
@suborb please if possible do not change standard behavior on targets that already have their escape/control codes.

If your generic driver provides some codes for all targets, will it not only break the code expecting the previous control codes, but it will also make the binary bigger because it won't rely any longer on bios/rom/basic routines, which come at zero cost for the binary/ram.

suborb, 2018-03-25:
Don't worry, it would be an alternate driver (in the same way ansi is for some targets)

It would follow the classic version (so 22,y,x etc)