z88dk/z88dk #660: [cpc] Unexpected error - Moreover the error does not specify the line

Fabrizio-Caruso, 2018-03-21:
My game now does not compile on the CPC target and produces an error that does not say much about the problem:
```
PROCESSING src/cpc/cpc_cpcrslib_graphics.c
ucpp -iquote"." -DZ80 -DCPC -D__CPC__ -DREDEFINED_CHARS -DSOUNDS -DFULL_GAME -D__CPC__ -DCPCRSLIB -DBETWEEN_LEVEL -DEND_SCREEN  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"c:/z88dk//include"    "src/cpc/cpc_cpcrslib_graphics.c" "D:\personal\cygwin64\tmp\zcc515B8.i2"
zpragma -sccz80 < "D:\personal\cygwin64\tmp\zcc515B8.i2" > "D:\personal\cygwin64\tmp\zcc515B8.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "D:\personal\cygwin64\tmp\zcc515B8.i"
make: *** [Makefile:551: cpc] Error 1
```

suborb, 2018-03-21:
I see this:

    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:235 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'
    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:241 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'
    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:246 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'
    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:255 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'
    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:260 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'
    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:271 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'
    sccz80:"src/cpc/cpc_cpcrslib_graphics.c" L:281 Error:Too many arguments to call to function 'cpc_PrintGphStrStdXY'

Running under lldb doesn't show sccz80 crashing, I'll check with valgrind on a linux machine later on

Fabrizio-Caruso, 2018-03-21:
This used to compile with just warnings. The problem is in cpcrslib but it was not stopping the compilation. If I remember well there is a mismatch between definition and declaration in cpcrslib.

suborb, 2018-03-21:
I changed it to an error in change d5baf3ec14cbfa3f27fabe5380b21980be274948 back in February.

* It's a pretty devastating error in a l->r calling environment - the parameters will be misplaced on the stack.
* It's deadly for __z88dk_callee - it will end up returning to a random address

Hence why it's now an error.

Fabrizio-Caruso, 2018-03-21:
How can I then compile my code that depends on cpcrslib?
Could you please provide an option to make this error a warning?

Fabrizio-Caruso, 2018-03-21:
@suborb I understand it should be an error but an option to have it as a warning is necessary to use cpcrslib with the CPC larget. At the moment the CPC target can only use cpcrslib to have some working graphics.

Fabrizio-Caruso, 2018-03-21:
The prototype is correct. The lib has something wrong but it works if it is called with the same parametwrs as the prototype.
Cpcrslib is no longer developed and I don't how to fix the .lib.

Fabrizio-Caruso, 2018-03-21:
@suborb as usual... you are right and I am wrong... please do not hate me for posting stupid things...

The problem seems fixed by a different prototype, which I had to guess:
```
cpc_PrintGphStrStdXY(unsigned char *col, char *cadena, char *x, char *y);
```

suborb, 2018-03-22:
No worries, I'm glad you found it and that I wasn't looking in the wrong place!

I think the prototype should probably be:

    void cpc_PrintGphStrStdXY(unsigned char col, char *cadena, unsigned char x, unsigned char y);

what you've got will work (probably with some warnings if you use -Wall) since sizeof(char *) == sizeof(int) and sccz80 passes chars as two bytes on the stack. The library function only pulls out the low byte of the word.

@zx70 was talking about pulling cpcrslib into the CPC port, if it's been orphaned it does need a maintainer, I'm not sure if it's still on his backlog though.

Fabrizio-Caruso, 2018-03-22:
@zx70 has done a great job (as usual) and has managed to integrate a good part of cpcrslib into the CPC port. I am his unofficial tester :-) It does work with maybe little glitches. This is a major improvement with respect to the previous situation of the CPC target because the CPC had no support for graphics and no support to display characters at a decent speed, at all.
It looks promissing. I can even compile my game without the external cpcrslib but, if I do so, I cannot easily redine characters (without modifying the compiler). We need to find a solution for it and for some minor graphics glitches.

zx70, 2018-03-22:
> I'm not sure if it's still on his backlog though.

Yes, I let a part of it in my backlog.

There's a fork of the library which recently evolved in a way that should be addressed in a way similar to what you did for the ANSI VT support (many optimized variants depend on the current graphics resolution, the chosen font, etc..).   The original way to address this situation was to have a skeleton (conceptually part of the library) provided by the programmer, which I don't like very much.  On the other side it is a pity to break the compatibility, so I haven't decided yet   :)

Fabrizio-Caruso, 2018-03-22:
I confirm that I see NO glitches in the embedded (partial) CPCRSLIB version in Z88DK.
It would be cool to have a way to redefine characters used in the multiple print functions.

Fabrizio-Caruso, 2018-03-22:
I have managed to get UDG with CPC using just the embedded (partial) version of CPCRSLIB.
THANKS STEFANO!
I have achieved this by just including the .asm file(s) with the character definitions. This file overrides the default definitions. This is cool!
Thanks to @zx70, we have graphics support in the CPC target. This opens many more possibilities! 
Even vanilla Z88DK is enough to produce games with some graphics!

Fabrizio-Caruso, 2018-03-23:
P.S.: This means no need to use cpcrslib unless some of the missing features in Z88DK are necessary.

zx70, 2018-03-23:
This is great news, basing only on what you see in your code, is something in my CPCRSLIB port still failing ?

Fabrizio-Caruso, 2018-03-23:
@zx70  evwrything's fine I am using one function. I will test more functions and let you guys know.

Fabrizio-Caruso, 2018-03-23:
P.S.: The very single function I use, alone, is already enough to display characters at a much higher speed than by the standard routines. Moreover, this single function can be used with user-defined characters. This means that Z88DK's CPC target can do efficient graphics without the need of any additional library.
@zx70 's port is already good. 
I will now try to use more functions and the more colorful mode 0 and see what I manage to get.