z88dk/z88dk #664: [m5][BASIC-I] getk exits to BASIC (with example)

Fabrizio-Caruso, 2018-03-24:
Under M5 with BASIC-I (it seems the most common BASIC),
the following program will just exit to BASIC:
```
#include <stdio.h>

void WAIT_PRESS(void)
{
	while(kbhit())
		getk();
	while(!kbhit())
	{ 
	}; 
	getk();
}
int main()
{
	getk();
	printf("Hello world!\n");	
	WAIT_PRESS();
	printf("Key pressed\n");	
	return 0;
}
```

Removing the getk in main, will produce Hello world and then exit once the next getk is reached.

suborb, 2018-03-25:
Can you send me the Roms you are using - if you’re using mess - I can’t find the f5 rom and mess refuses to start up without it.

Fabrizio-Caruso, 2018-03-25:
You need both the bios roms and the basic cartridge rom.
[m5.zip](https://github.com/z88dk/z88dk/files/1844918/m5.zip)
[m5p.zip](https://github.com/z88dk/z88dk/files/1844919/m5p.zip)
[Home Basic (Japan) (SC-3000).zip](https://github.com/z88dk/z88dk/files/1844920/Home.Basic.Japan.SC-3000.zip)

suborb, 2018-03-25:
Thanks for that I've now got both mame and emu5 up and running.

emu5 is bit easier to debug with I have to say.

I've got fgetc_cons() working and getk() is working by itself, just in your example it's not.

At this point, it looks like reading the keyboard presents the L cursor which is a bit weird.

suborb, 2018-03-25:
It looks like all the info on the m5 has vanished from the internet - about the only list of routines available is the one in z88dk (m5bios.def)

Anyway, getk() now correctly scans the keyboard rather than waits for a keypress.

Your example code is a bit wrong: kbhit() should be followed by a getch() to retrieve the pressed key.

I couldn't reproduce your crash whilst reading the keyboard, but at the end of the run, it looks like basic resets - this may be design in the BASIC, I'm honestly not certain.

commit 30837930 (2018-03-25): Fix #664: getk() on the m5 now just polls the keyboard rather than
waiting for a keypress.

fgetc_cons() consumes from the keyboard buffer.
closed 2018-03-25 by suborb