z88dk/z88dk #666: [classic] Create a standard console driver for memory mapped character targets

suborb, 2018-03-25:
So that there's a standard set of terminal codes available across many machines (so a lightweight ANSI engine)

Thinking of trs80, z1013, possibly zx80, zx81, jupiter ace?

Fabrizio-Caruso, 2018-03-25:
This is an idea of how code behind the scenes is shared among similar targets.
For targets with indirect video memory access (e.g. VDP) a common VPEEK and VPOKE implementation would be nice. If you hace these command, you can extend the driver to these indirect cases.

What would be really nice is to have a common API for ALL targets including those with indirect video memory access. Simple conio macros could be used.

Fabrizio-Caruso, 2018-03-25:
@suborb, great news!
If this is coded in C, it is easy to make it to also cover indirect video memory access (e.g., VDP)
by coding in a sort of "polymorphic" way:
Just pass function pointers for memory read and write functions.
Use standard POKE, PEEK for direct access and VPOKE, VPEEK for VDP targets.
This can be very elegant.

zx70, 2018-03-30:
I just added a tiny console implementation, it is meant for low resolutions (e.g. vz200), where some text is necessary together with graphics without the memory waste of the X11 lib.
It is based on plot/unplot and works also on targets with very basic graphics support.

Thanks to Dom its insertion is quite easy:
-pragma-redirect=fputc_cons=putc4x6 

x_4x6 and y_4x6 can be referenced to position your "text" at a specific PIXEL position, text scrolling is not supported.

zx70, 2018-04-07:
a simpler console driver based on f_ansi_char is easy to do, we need to decide how much simple we are willing to keep it and which standard to adopt.
chr(12) to clear screen, gotoxy(), chr(8) for bs ...   scrolling text or not?

suborb, 2018-04-10:
I was really thinking about character mapped displays - where we just have to put a byte at the right point in memory for it to be displayed at the native resolution.

I'm sort of proposing something like this:

[fputc_generic.asm.txt](https://github.com/z88dk/z88dk/files/1896426/fputc_generic.asm.txt)

It's based on the classic ZX codes:

```
;  8 = backspace (left)
;  9 = right
; 11 = up
; 13 = down
;  4,[0|1] = enable/disable vertical scroll
; 10 = line feed
; 12 = cls
; 22,y,x = Move to position
```

And needs the following to be implemented for each target:

```
defc CONSOLE_ROWS
defc CONSOLE_COLUMNS

generic_console_cls - clear the screen

generic_console_scrollup - scroll the screen up

Must preserve: bc, de

generic_console_printc  - print the character

Entry:  c = x
            b = y
            a = character to print
            e = 0/1 (1 = raw mode, don't translate character)
```

There's an additional function: void fputc_generic_setmode(char raw) __z88dk_fastcall which will cook into raw mode/translate mode. This way graphics can be printed on non-ASCII machines.

I've not tried out the code (and I think y and x are mixed up in a couple of places) and the names of a bit iffy, but it comes to about 220 or so bytes which is probably acceptable.

zx70, 2018-04-11:
Got it, this could be a very comfortable hook e.g. for a simple conio implementation.    Would you mind to go ahead with a prototype ?

suborb, 2018-04-11:
I'll stick something up in a branch over the next few days, be good to get some feedback from @Fabrizio-Caruso and RobertK before merging.

suborb, 2018-04-11:
I think there's enough there to prove the concept, in that branch compile with -pragma-redirect:fputc_cons=fputc_cons_generic to use it.

Before merge I want to come up with a better name.

zx70, 2018-04-18:
there is plenty of potential targets to add, do you think I can just add my guesses to your list ?   I honestly prefer to try  and extend to them if it works.

closed 2018-04-25 by suborb