z88dk/z88dk #667: [classic] Various low res plotting issues

suborb, 2018-03-26:
As reported on the forum: https://www.z88dk.org/forum/viewtopic.php?id=10250

Problematic ports are below, @zx70 can you help?

 === Jupiter Ace ===
   zcc +ace -vn -create-app plottest.c -o pt.bin
   Note: compiles and runs without error, but plotted points 
   cannot be seen. Point() function returns inverse values.
   Load the program on the Jupiter Ace by typing:
   0 0 bload pt.bin
   
   === VZ200 / Laser 210 / Laser 110 ===
   zcc +vz -lm -o plottest_vz200.vz plottest.c 
   Note: plot() function outputs characters instead of pixels. Switching to 
   graphics mode required? "The graphics mode is automatically switched on 
   by calling “clg” or printing CHR$ 12 or calling the vz_mode() function."
   Point() function works correctly.
   
   === TRS-80 === (almost ok)
   zcc  +trs80 -lndos -lm -create-app -o plottest_trs80.bin -subtype=disk plottest.c
   Note: use a TRS-80 Model III, type "system" to exit Basic and run the .cmd file.
   Clearing the screen does not work with 1.99B, since nightly build 2018-03-24 
   this should work as well. Compiled program runs very unstable on the M.A.M.E. 
   emulator (version 0.170), many attempts are needed to get it running correctly.
      
   === Philips VG5000 ===
   zcc +vg5k -create-app -lm plottest.c -o plottest_vg5000
   Note: Everything works, except for that plot() function 
   outputs " characters instead of pixels.
   Use the DCVG5K emulator (dcvg5k.free.fr) for testing
   Attach the .k7 cassette file and type CLOAD to load the program
   (note that the VG5000 has a French AZERTY keyboard so on most 
   other keyboard layouts you have to type CLOQD)
   
   === Mattel Aquarius ===
   zcc +aquarius -lm -create-app plottest.c -o plottest_aquarius
   Note: Everything works, except for that plotted pixels are not 
   all the same size. In the plot test the second pixel looks ok 
   while the first and third are too flat.
   Use the "Virtual Aquarius" emulator for testing. Loading 
   instructions from the z88dk wiki:
    from within the emulator, load the loader: cload
    press ENTER twice, then play the “_plottest_aquarius.caq” cassette file (loader)
    at the “OK” prompt, type RUN <ENTER>
    press ENTER twice, then play the “plottest_aquarius.caq” cassette file
    wait… then have fun !

   === EG2000 ===
   zcc  +trs80 -lndos -lm -create-app -subtype=eg2000disk -o plottest_eg2000.bin plottest.c   
   Note: plotted points cannot be seen, everything else is ok.
   Use the Genieous emulator (gaia.atilia.eu) for testing. See the TRS-80 
   loading instructions above (use "Autostart file" after entering SYSTEM)
   
   === Sharp MZ-700 ===
   zcc +mz -lm -o adventure -create-app plottest.c -o plottest_sharpmz
   Note: clearing the screen does not work, everything else is ok.
   In M.A.M.E. attach the .mzt cassette file, then press L and Enter 
   to initiate loading, then use M.A.M.E.'s tape control to start the tape.

zx70, 2018-03-26:
yup, I'm here  :)

zx70, 2018-03-27:
Too flat pixels on the aquarius are expected, there is still something to be explored by inverting the color attributes and recycling some better shaped symbol.
As for the J ace, there is also the possibility to provide two resolutions on two parallel libraries

zx70, 2018-03-28:
new 80x48 option for the aquarius (also here the pixels are not all of the same thickness).
different character (thicker) chosen in 80x72 mode for the bottom-right dot which was far too odd

zx70, 2018-03-28:
The laser 200 works as expected, it is not possible to mix text and graphics, any text output (e.g. an error message in BASIC) will exit from the graphics page restoring the text mode. 
I  strongly tempted to provide a tiny simplified font and optionally replace the fput_cons call with a very basic plotpixel driven function, obviously outputting chr(12) would trigger cgl().

zx70, 2018-03-30:
[image]

It is now possible to print something also on targets not permitting to mix text and graphics :)

    #pragma redirect fputc_cons = putc4x6

..or, at compile time:   -pragma-redirect=fputc_cons=putc4x6

Should work on any target excluding the ones featuring "wide resolutions" (> 256 horizontal pixels).

zx70, 2018-03-30:
I succeeded to run the test program on the vg5k.   In any case I'm inserting a new option (-clib=graphics) to support the new 4x6 pseudo-text mode.

zx70, 2018-04-04:
[image]

CGENIE result with the new pseudo-text option:
-pragma-redirect=fputc_cons=putc4x6

closed 2018-04-04 by zx70

suborb, 2018-04-04:
@zx70 is there a way to convert the 64 column ZX fonts into a font that can be used with this screen printer?