z88dk/z88dk #669: z80nm: add ability to list used sections and reassign sections

aralbrec, 2018-03-28:
There's a need to reassign sections in object files and libraries.

One use case is the incorporation of 3rd party library code into a project spread across memory banks.  The project probably has a custom memory map and you'd like to place the 3rd party code in a specific point in memory.  That can be done if the sections the 3rd party code is assigned to can be changed to what is needed in the project.

A second use case is to compensate for zsdcc's lack of section reassignment for bss (may be solved in later versions of sdcc) and data variables.  At the moment zsdcc is unable to change which section bss and data variables are compiled to, forcing these variables to be created in asm.  But with this section reassignment feature in z80nm, the object files could be manipulated to change section assignment for those variables.

I'll leave this open and see if someone wants to take it up before I have some time.

aralbrec, 2018-03-28:
Perhaps an ability to modify label names as well might be useful, perhaps via regular expression.

pauloscustodio, 2018-03-28:
If I understood well, the need is to rename sections and labels inside object and library files without merging. Merging two sections into one requires a new linker run.

I don't like the usage of z80nm for this - it is a viewer, and has built-in code to parse all the different object file formats out there. 

Maybe a new z80link could do this, but only for the latest object file format?

pauloscustodio, 2018-03-29:
> Yes and I imagine it would be ok if two different section names in a particular object file were renamed to
> the same section. They just wouldn't be merged together.

I'm not sure, will have to check how the linker behaves. Addresses are computed as section_base+offset, so the second section with the same name would probably have the addresses wrong.

pauloscustodio, 2018-03-30:
zobjcopy expected soon...

pauloscustodio, 2018-03-31:
> Yes and I imagine it would be ok if two different section names in a particular object file were renamed
> to the same section. They just wouldn't be merged together.

This will not work with the current object file version (11) because patch expressions and symbols only refer to the section they belong to by name; if two sections are merged into one, then all the patch and symbols will refer to the first of the list of sections merged, producing incorrect links.

The solution will be to associate the expressions and symbols to each of the section blocks instead of one global set per object file. This is also in the path of getting an object file to contain multiple independent modules.

pauloscustodio, 2018-04-15:
@aralbrec, please check the feature/zobjcopy branch and the wiki https://github.com/z88dk/z88dk/wiki/Tool-zobjcopy and let me know of any issues before I merge.

aralbrec, 2018-04-15:
> Code sections are merged, if appropriate, fixing the patch addresses of expressions and label values accordingly.

Does this mean I can rename two different sections to the same one and things will still be ok?

`zobjcopy ff.o --section ^code_compiler=code_fatfs --section ^bss_compiler=code_fatfs new.o`

I still see two different `code_fatfs` sections that seem to have the same offset:

```
  Code: 5 bytes (section code_fatfs)
    C $0000: 00 00 00 00 00
  Code: 0 bytes (section IGNORE)
  Code: 0 bytes (section code_crt_init)
  Code: 18351 bytes (section code_fatfs)
    C $0000: CD 00 00 11 04 00 CD 00 00 0A D3 EB 03 1B 7A B3
```

pauloscustodio, 2018-04-15:
> Does this mean I can rename two different sections to the same one and things will still be ok?

Yes. I did not cover one case, though: if the renamed-to-section already exists, you'll end up with two sections with that name.

I'll add a test case and a fix.

aralbrec, 2018-04-20:
A couple of errors:

```
zobjcopy ff.o --section ^code_compiler=code_fatfs --section ^bss_compiler=code_fatfs new.o
error: section '' not found in file 'ff.o'

zobjcopy -l ff.o
error: file 'ff.o' version 110 not supported
```

The library version was bumped so I will try again after rebuilding but I thought `zobjcopy` would be able to read the older library format and update to the current library format?

aralbrec, 2018-04-20:
After rebuilding:

```
zobjcopy ff.o --section ^code_compiler=code_fatfs --section ^bss_compiler=code_fatfs new.o
error: file 'ff.o' version 118 not supported
```

```
zobjcopy -l ff.o

Object  file ff.o at $0000: Z80RMF11©╦6☺¾UÄ▒Þ¨§
  Name: ff
  Section bss_compiler: 5 bytes
    C $0000: 00 00 00 00 00
  Section IGNORE: 0 bytes
  Section code_crt_init: 0 bytes
  Section code_compiler: 18324 bytes
    C $0000: CD 00 00 11 04 00 CD 00 00 0A D3 EB 03 1B 7A B3
    C $0010: 20 F7 DD E1 C9 CD 00 00 11 04 00 CD 00 00 DB EB
...
...
    C $41B0: 50 C5 DD 7E 04 F5 33 D5 CD 00 00 F1 33 CD 00 00
    C $41C0: DD F9 DD E1 C9 CD 00 00 21 B6 FF 39 F9 21 00 00
error: section '' not found in file 'ff.o'
    C $41D0: 39 5D 54 E5 DD 6E 06 DD 66 07 E5 D5 CD 00 00 F1
    C $41E0: F1 C1 DD 71 FE DD 70 FF DD 5E 04 DD 56 05 1A B7
...
...
    C $4780: 33 D1 C1 18 DD DD 4E EB DD 46 EC C5 CD 00 00 DD
    C $4790: F9 DD E1 C9
  Section data_compiler: 1 bytes
    C $0000: 01
  Section rodata_compiler: 15 bytes
    C $0000: 22 2A 2B 2C 3A 3B 3C 3D 3E 3F 5B 5D 7C 7F 00
  Symbols:
```

pauloscustodio, 2018-04-20:
The library version was not bumped. There seems to be a alignment error while reading the binary file that I did not have in my tests.

Is your architecture 64 bits?
Where is the source of ff.o so that I can try to reproduce the failure?

Thanks.

pauloscustodio, 2018-04-20:
I managed to reproduce the error in Ubuntu 64-bits:

    error: file 'test.o' version 110 not supported

It must be a 32 vs 64 bits bug. I'll investigate.

aralbrec, 2018-04-20:
It's getting further but I am still getting an error that aborts the operation:

```
zobjcopy ff.o --section ^code_compiler=code_fatfs --section ^bss_compiler=code_fatfs new.o
error: section '' not found in file 'ff.o'
```

A `zobjcopy -l ff.o` lists the object file contents but it also comes up with the same error.

`ff.o` can be found in:
https://drive.google.com/file/d/1VMNzaMA5PhLjZr4bTjhIv9pBr6uB7-H2/view?usp=sharing

pauloscustodio, 2018-04-20:
Fixed the case of a missing "" section. Please try again and let me know of further issues.

pauloscustodio, 2018-04-22:
zobjcopy basic features finished.

closed 2018-04-22 by pauloscustodio