z88dk/z88dk #672: [ace] recent changes has broken conio (extra characters displayed)

Fabrizio-Caruso, 2018-03-29:
The ACE target had a working conio implementation, which is no longer working correctly.

Extra characters are displayed.
[image]

zx70, 2018-03-30:
I just told Fabrizio the ACE target was let untouched, but it is not totally true..
While solving #667 I had to revive the base_graphics lablel I committed the single row number https://github.com/z88dk/z88dk/commit/a1a6ba4ebbfb11f1efc1be6f395fedc060734387 .

Fabrizio,  I can't imagine why this could compromise a thing but please try by removing from  lib/target/ace/classic/ace_crt0.asm the block of rows 127..129:

    ELSE

	    ld   hl,$2400
	    ld   (base_graphics),hl

Fabrizio-Caruso, 2018-03-30:
The culprit was MY code (sort of):
```
-               #define cputc(c) printf("\x11%c%c",COLOR_BLACK,c)
+              #define cputc(c) fputc_cons(c)
```
which, used to work even though I was display a character with 
`printf("\x11%c%c",COLOR_BLACK,c) `

I have fixed it by just doing a straightforward ` fputc_cons(c)`.

closed 2018-03-30 by Fabrizio-Caruso