z88dk/z88dk #673: z80asm: MODULE directive cannot accept names that coincide with register or flag names

aralbrec, 2018-03-30:
I found this while trying to compile a program "a.c" with sccz80.   sccz80 will generate a "MODULE a" directive in its output asm which z80asm will flag as error.

a.asm
[code]
MODULE a

PUBLIC foo

foo:

   ld hl,10
   ret
[/code]

`z80asm a.asm`
Error at file 'a.asm' line 1: syntax error
1 errors occurred during assembly

Change `MODULE a` to `MODULE g`

`z80asm a.asm` works fine.

I find I can't use register or flag names in the MODULE string.

pauloscustodio, 2018-03-30:
The current lexer does not accept identifier names that might appear as operands, i.e. register and flag names.

Is this really a bug?

aralbrec, 2018-03-30:
Right now you can't have source files with names "a.c", "b.c", etc so that's a bug someplace.  If not in z80asm then sccz80 should be changed to so that it doesn't generate module names like "a", "b", etc.  It could keep the file extension so the module names are "a.c", "b.c".

[zcc](https://github.com/z88dk/z88dk/blob/master/src/zcc/zcc.c#L1328) is automatically inserting module names into files that are connected to the input filenames that keep the extensions except for input *.asm files in which case the ".asm" is stripped.  So I expect "a.asm", "b.asm" etc to have a similar problem when processed by zcc.  That special treatment for *.asm in zcc should probably be removed.

sccz80 is overriding the module name generated by zcc because it inserts its own that appears after the zcc one.  We could remove the module name generation from sccz80 but long term that's the opposite of what will be needed because down the road "MODULE" will be used to block out separate sections within a file so the c compiler will have to generate module directives at that time.

So maybe we fix that one issue with zcc and modify the module names sccz80 is generating?
@suborb

pauloscustodio, 2018-03-30:
Let me check if I can tweak the lexer to ignore keywords after Module.

suborb, 2018-03-30:
Whilst you’re there....there’s code on sccz80 to map away initial numbers as well which also caused problems in the past. 

However, it depends if we’re ever going to be treating module names as a symbol in the future.

aralbrec, 2018-03-30:
The problems should be sorted now.  In a test using input files "a.c" and "b.c" I noticed sccz80 is creating module names "a_cH" and "b_cH" - I'm not sure if that's intentional?

I suppose all that remains is to decide if MODULE takes a string or an identifier.

pauloscustodio, 2018-03-30:
I think it should be an identifier, just in case we want to create some module-dependent symbols (e.g. MODULE_size)

suborb, 2018-03-30:
> The problems should be sorted now. In a test using input files "a.c" and "b.c" I noticed sccz80 is creating module names "a_cH" and "b_cH" - I'm not sure if that's intentional?

They should be "a_c" and "b_c" - any non-alpha initial letter is replaced with an underscore, and subsequent non-alphanumeric are replaced with an underscore. I think z80asm had an issue with "." in the name at one point as well.

I'm not sure where the H is coming from, I don't see it on my machine - what's in the `#line` directive? I have:

    #line 1 "a.c"

    int main() {
    }

pauloscustodio, 2018-03-30:
Closed. Module and Section should now be able to be any identifier name.
A dot (.) is not allowed in identifiers, therefore a module cannot have a name with a dot.

closed 2018-03-30 by pauloscustodio

suborb, 2018-03-30:
@aralbrec sccz80 gets its details from the .i file so what's the output of a run with -E?

aralbrec, 2018-03-31:
zcc +zx -vn -E a.c

```
#line 1 "a.c"
int trace;

void main(void)
{
}
```

aralbrec, 2018-03-31:
I'm not sure why I can't process the .i file with sccz80:

zcc +zx -vn -E a.c
`creates a.c.i`

zcc +zx -v -a a.c.i
```
PROCESSING a.c.i
sccz80  -ext=opt -mz80 -standard-escape-chars   "a.c.i"
copt "c:\Projects\z80\z88dk\z88dk\lib\config\..\..\/lib/z80rules.9" < "C:\Users\
alvin\AppData\Local\Temp\zccD7292.opt" > "C:\Users\alvin\AppData\Local\Temp\zccD
7292.op1"
The system cannot find the file specified.
```

With zsdcc:

zcc +zx -vn -E -clib=sdcc_iy a.c
`creates a.c.i`

zcc +zx -v -a -clib=sdcc_iy a.c.i
`creates a.c.i.asm`

suborb, 2018-03-31:
Hmmm, sccz80 doesn't support the -o option so the a.c.i gets transformed to a.c.asm in the current directory.

aralbrec, 2018-04-01:
Yeah that is fixed but the module names are still strange.

a.c
```
int trace;

void main(void)
{
}
```

zcc +zx -vn -E a.c

a.c.i
```
#line 1 "a.c"
int trace;

void main(void)
{
}
```

zcc +zx -v -a a.c.i

a.c.i.asm
```
;* * * * *  Small-C/Plus z88dk * * * * *
;  Version: MSC build-20180401
;
;	Reconstructed for z80 Module Assembler
;
;	Module compile time: Sun Apr  1 08:54:26 2018

	MODULE	a_cs
...
```
It looks like an off by one buffer error.

Similarly if I rename the file "foobar.c":

```
;* * * * *  Small-C/Plus z88dk * * * * *
;  Version: MSC build-20180401
;
;	Reconstructed for z80 Module Assembler
;
;	Module compile time: Sun Apr  1 08:57:02 2018

	MODULE	foobar_c_
```
There's an extra trailing underscore.

commit 68d62941 (2018-03-30): Fix #673 : allow register and flag names in the SECTION and MODULE directives