z88dk/z88dk #674: [abc800] appmake does not work (the doc shows -create-app in the example)

Fabrizio-Caruso, 2018-03-31:
appmake does not work (the doc shows -create-app in the example)

zx70, 2018-03-31:
it is a mistake, no abc800 specific appmake is available

zx70, 2018-03-31:
I don't remember honesty how the console support was tested.    Emulators were still incomplete,  I could have injected binary code somewhere. 
by the way i read an abc800 manual stating that the ABC 80 programs were loadable from the cassette interface,   I will investigate.

Fabrizio-Caruso, 2018-04-03:
@zx70 I cannot see any option in MESS/MAME to load a binary directly into memory.
Maybe you used a different emulator to do your tests.

Fabrizio-Caruso, 2018-08-28:
I still have no found a way to load the raw binary.
Maybe there is a complicated way to inject into memory in Mame.
Even a stupid BASIC loader would be useful at least for testing the target.

Fabrizio-Caruso, 2018-08-28:
@zx70  I just found:
http://www.lysator.liu.se/~exon/archive/ABCWin2/
where in ABCTools you find a tool to convert files into WAV format. It is not clear to me, which source format is expected. If it is raw binary, we may have found a nice solution for both ABC80 and ABC800.