z88dk/z88dk #678: sccz80: incorrect 32-bit equality comparison in compound if

aralbrec, 2018-04-05:
zzz.c
```
#include <stdint.h>

static uint8_t  version;
static uint32_t mem_size;

void test(uint32_t ptr)
{
    if ((version < 4) && (mem_size == 0x10000))
    {
        ptr = ptr & 0xFFFF;
    }
}
```

zcc +zx -vn -a zzz.c

```
._test
	ld	hl,(_version)
	ld	h,0
	ld	a,l
	sub	4
	jp	nc,i_3
	ld	hl,(_mem_size)
	ld	de,(_mem_size+2)
	push	de
	push	hl
	ld	hl,0	;const
	ld	de,1
	call	l_long_eq
	jp	nc,i_3
	ld	hl,1	;const
	jr	i_4
.i_3
	ld	hl,0	;const
.i_4
	ld	a,h
	or	l
	or	d
	or	e
	or	d
	or	e
	jp	z,i_2
	ld	hl,2	;const
	add	hl,sp
	push	hl
	ld	hl,4	;const
	add	hl,sp
	call	l_glong
	ld	de,0
	pop	bc
	call	l_plong
.i_2
	ret
```

If the test of version fails it actually jumps into code that looks like is part of the following equality compare.  The equality compare uses "l_long_eq" but then also tries to set HL to 0/1 for false/true when "l_long_eq" already does this.  This is following by testing HL for true/false but instead DEHL is tested which gives an incorrect result.

suborb, 2018-04-05:
@aralbrec  I'm unable to compile that snippet with zsdcc - I'm getting "conditional flow changed by optimizer: so said EVELYN the modified DOG"

aralbrec, 2018-04-05:
Yeah it's noticed `ptr` is a local variable and it just eliminates all the code because `ptr` can't be observed outside the function.

This will work:

```
#include <stdint.h>

static uint8_t  version;
static uint32_t mem_size;

uint32_t test(uint32_t ptr)
{
    if ((version < 4) && (mem_size == 0x10000))
    {
        ptr = ptr & 0xFFFF;
    }

   return ptr;
}
```

aralbrec, 2018-04-05:
> I'm not going to merge that PR until after the weekend so if you want to give it a try out..

You still need to check the PR?  Or should I merge it if it passes?

suborb, 2018-04-06:
It should be good. I’m a bit unhappy about the %% in the Copt eval extension. 

But if the bulk of it is good for you then merge - I’m away from a computer for a few days.

aralbrec, 2018-04-06:
The fix looks ok.  I still have a different problem somewhere to do with 32-bit code but hopefully will find it soon.

commit fd92526b (2018-04-06): Merge pull request #679 from z88dk/bugfix_compounds

Fix #678 Bugfix compound conditions, add RPN calculator to copt.
closed 2018-04-06 by aralbrec