z88dk/z88dk #683: [clg][KC] clg() produces freezes and random glitches on most KC sub-models

Fabrizio-Caruso, 2018-04-08:
Hi

I have successfully managed to use the sprite engine with KC! 
I have realized that clg() produces random wrong effects and freezes on most KC sub-models at least when used inside CROSS CHASE (tested with jkcemu):
- on KC/4 -> it breaks the input
- on KC/5 -> on the second execution it locks the computer
Similar problems and/or graphics corruptions are also visibile on these two and other sub-models.

zx70, 2018-04-10:
I can't really explain the impact of clg(), it is a simple ROM call (like the one used all the time by pixeladdress).
In any case I replaced it with a dumb loop clearing the graphics memory (top-left position identified by pixeladdress, then zeroing the next 10240 bytes).
I'm not altering the color attributes to limit the impact, let me know if the tomorrow's build will solve.

zx70, 2018-04-10:
I think I'm getting closer to the root cause, if it is the way I think (ix register affecting the os stability) the last fix won't help much.

zx70, 2018-04-10:
so, googling with the right keywords it is easy to confirm..
https://sourceforge.net/p/sdcc/mailman/message/33036952/

suborb, 2018-04-10:
Aha, lets make kc an ixiy target then  - I presume iy is available?

zx70, 2018-04-11:
My first experiment with the ixiy workaround and the iy library failed, the resulting stability looks worse than with the existing interrupt wrappers protection.   I'm investigating on possible mistakes,

Fabrizio-Caruso, 2018-04-11:
@zx70, some initial tests (only KC 85/5) show that the "fix" has worked by probably avoiding by sheer chance the use of a forbidden register.
The root cause is still there, though, if I understand well.

Fabrizio-Caruso, 2018-04-11:
@zx70 
The new clg() "fixes" all 5 sub-models:
HC 900 *
KC85/2 *
KC85/3 *
KC85/4
KC85/5
The first 3 models (*) have a little glitch: the cursor remains visible on the screen.

zx70, 2018-04-11:
I'm afraid it is just a matter of probability,  in example dstar.c starts correctly clearing the screen perfectly.
Then the game field is drawn and the ball moves under the keyboard control as expected.
On any next clg() the picture gets garbled, probably after the getk() usage random memory areas are corrupted, or some kind of overflow occurs.   In the "CAOS reference" document it is stated:

Restrictions
• do not modify the IX register, this is needed as base pointer to system memory variables by keyboard and cassette loading interrupts
• the only supported interrupt mode is IM2

suborb, 2018-08-25:
I was looking at direct keyboard input for the kc to see if we can bypass the problems with ix.

Unfortunately, the jkcemu emulator actually uses the value of ix when a key is pressed/released so we can't avoid it all unfortunately.

feilipu, 2022-10-18:
Closed by #2092.

closed 2022-10-18 by feilipu