z88dk/z88dk #685: sam coupe

rogerjowett, 2018-04-10:
https://www.youtube.com/watch?v=zsjTpFR0oYQ

please can it emulate the sam coupe yet?

zx70, 2018-04-10:
Sorry I don't your post.   The link to the video shows (probably) a sequence of pictures loaded by the SAM Coupè to provide a "video stream".
What are you exactly proposing/asking to the z88dk team ?

rogerjowett, 2018-04-10:
well i thought the z88dk was a z80 emulator is that not right?

zx70, 2018-04-10:
not exactly, z88dk is a developemnent kit for the z80 (including a c compiler, an assembler and libraries) but it is targeted to extermal computers (or emulators).

closed 2018-04-10 by zx70

rogerjowett, 2018-04-11:
there are a lot of emulators for the zx spectrum about but on ly one sofar
is trying to model the uhf tv output

https://www.aptanet.org/eightyone/

https://sourceforge.net/projects/eightyone-sinclair-emulator/

suborb, 2018-04-11:
Roger, you'll probably get a better response if you take your thoughts to https://worldofspectrum.org/forums/categories/emulators 

Thanks

rogerjowett, 2018-04-11:
> i cant they banned me
> ok sorry

- Sorry, permission denied.