z88dk/z88dk #686: [ace] Create in_Inkey and friends implementation

suborb, 2018-04-11:
The ace keyboard is very similar to the ZX keyboards:

http://www.forums1.jupiter-ace.co.uk/prog_keyboardread.html

[image]

zx70, 2018-04-11:
joystick() ready

zx70, 2018-04-11:
Further reference:
http://www.forums1.jupiter-ace.co.uk/hardware_ace_inteface.html
Often suggested: "multipress on keys 1 to 5", but I kept the number keys mapping as for the "Sinclair 1"/"Sinclair 2" joysticks.   Befor adjusting the sequence I'd like to find documentation on whether a Kempston interface could be adapted in a similar way to what someone did between the ZX81 and the Spectrum edge connectors.

closed 2018-05-17 by suborb