z88dk/z88dk #687: [cryptic compile error]  Compilation error does not seem to tell much

Fabrizio-Caruso, 2018-04-11:
Once again I am not able to understand some cryptic compilation errors.
Am I doing something wrong? How should I investigate such errors?

After some successful compilation steps I get a cryptic error:

```
...
PROCESSING rm.asm.m4
m4 -I "c:\Z88DK\lib\config\..\..\\lib\target\nc\classic" -I "c:\Z88DK\lib\The system cannot find the file specified.
config\..\..\/src/m4"  --define=__STARTUP=1  < "rm.asm.m4" > "C:\cygwin\tmp\zcc4A4314"
make: *** [Makefile:1177: nc100_sprites_light] Error 1
```

suborb, 2018-04-11:
I think you’ve got the wrong file name somewhere. Basically a file doesn’t exist and for some reason it’s being assumed to be an m4 file and failing. 

We saw this in another issue - I think on the forum.

Fabrizio-Caruso, 2018-04-11:
@suborb you are right as usual... Sorry for reporting this

closed 2018-04-11 by Fabrizio-Caruso