z88dk/z88dk #688: [wrong compilation error] error at file 'foo' line x, shows wrong line number

Fabrizio-Caruso, 2018-04-11:
I was getting this error message:
```
$ PROCESSING CRT

z80asm --cpu=z80 -b -d  -o"build/LIGHT_nc100_sprites.bin" -m -s -L.   -L"c:\Z88DK\lib\config\..\..\/lib/clibs" -I"c:\Z88DK\lib\config\..\..\/lib" -igfxnc100 -inc_clib -indos -i"z80_crt0"     "@C:\cygwin\tmp\zcc6DC515.lst"
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\nc\classic\nc100_crt0.asm:
Error at file 'src/text.c' line 55: symbol '_gotoxy' not defined
                   ^ ----
Error at file 'src/text.c' line 61: symbol '_putch' not defined
                   ^ ---- #define CPC_TEXT_WHITE 3
-bash: PROCESSING: command not found
```

but line 55 in src/text.c is empty

and line 61 is just an innocent define derictive

I was able to guess the error, which was further down in the same file src/text.c

suborb, 2018-04-11:
That's showing the C filename, but the assembler line number. It's tied up with #611

suborb, 2020-07-06:
This got revisited in #1510 - I ended up emitting C_LINE from sccz80 to get the line number closer.

closed 2020-07-06 by suborb