z88dk/z88dk #689: zsdcc:  code generated for c fastcall function with locals on the stack wipes out input param

aralbrec, 2018-04-12:
Example function from magnetic scrolls:

```
void char_out(type8 c) __z88dk_fastcall
{
    static type8 big = 0, period = 0, pipe = 0;

    if (c == 0xff)
    {
        big = 1;
        return;
    }
    c &= 0x7f;
    if (read_reg(3, 0))
...
```

Declared fastcall, the input parameter is in register `L`.  However `L` is wiped out by the function prologue that reserves spaces on the stack for local variables:

```
4656  0A02              ;	---------------------------------
4657  0A02              ; Function char_out
4658  0A02              ; ---------------------------------
4659  0A02              ;	Register assignment might be sub-optimal.
4660  0A02              ; Stack space usage: 8 bytes.
4661  0A02              _char_out:
4662  0A02  DD E5       	push	ix
4663  0A04  DD 21 00 00 	ld	ix,0
4664  0A08  DD 39       	add	ix,sp
4665  0A0A  21 F8 FF    	ld	hl, -8   ;;; wipes out input parameter
4666  0A0D  39          	add	hl, sp
4667  0A0E  F9          	ld	sp, hl
4670  0A0F  7D          	ld	a,l
4672  0A10  3C          	inc	a
...

Must check if later versions of sdcc correct this problem.

spth, 2018-05-11:
This looks like [upstream SDCC bug #2580](https://sourceforge.net/p/sdcc/bugs/2580/), which was fixed in upstream SDCC 7 months ago.

Philipp

spth, 2018-05-11:
Am 11.05.2018 um 19:24 schrieb aralbrec:
> I will have another go at updating to the current sdcc soon. We're
> targetting a new z80 cpu variant for the zx next and will have to add a
> few instructions for the peepholer.
> 
> Are you interested at all in making some tweaks to the code generator to
> target this z80 variant?
> New instructions here <https://github.com/z88dk/z88dk/issues/312>. Be
> warned this is still fluid.

I'm open to adding support for another z80-related target with
additional instructions in SDCC. But:
1) Someone else has to implement support in the sdas assembler
2) Someone else has to implement support in the ucsim simulator

Philipp

pauloscustodio, 2018-05-11:
I can work on sdas if this has a go.

aralbrec, 2018-05-12:
Looking at the [ucsim source](https://sourceforge.net/p/sdcc/code/HEAD/tree/trunk/sdcc/sim/ucsim/z80.src/), I think it probably isn't hard to add the z80-zxn cpu type.  There is already an exception for the z180 in the z80 simulation so adding another for the z80-zxn should be straightforward.  The rabbits have a separate cpu module.  Even so, maybe asking drdani if he would like to make the changes would work - he was very active in the ucsim module a few months ago.  I'll fire off an email and see if he's interested if not, maybe I can make the changes and have him review.

Changing sdas may be a taller order.  Maybe have a look to see how hard it would be?

pauloscustodio, 2018-05-12:
> Changing sdas may be a taller order - it may affect the linker and object format. IIRC the object files
> are text files. Maybe have a look to see how hard it would be?

I don't think we need to change the object format. From a short read in supports both little- and big-endian expressions (the latter required for the copper instructions).

I think we can include zxn in the same was as z180: with a ".zxn" directive.

The first step seems to be harder - getting sdcc to compile in Windows. msys stops with missing symbols (NO_ERROR, ERROR_INVALID_HANDLE); the Visual Studio solution files complain a missing header (SDCCy.h)

Any pointers to help?

spth, 2018-05-12:
> Any pointers to help?

I suggest to ask on the sdcc-user list. AFAIK, it is mostly Maarten that takes care of the build of SDCC on Windows. I mostly use GNU/Linux (and once in a while check that SDCC compiles on OpenBSD and Hurd, and try to fix any issues that come up there).

Philipp

aralbrec, 2018-05-13:
I compile in windows with vs2015 but there are a few small changes that have to be made.

SDCCy.h probably means the version of bison that you're running is not compatible with the version the sdcc maintainers are running.  I had to change src/SDCC.y like so:

```
===================================================================
--- src/SDCC.y	(revision 9958)
+++ src/SDCC.y	(working copy)
@@ -182,7 +182,7 @@
           allocVariables ($1);
           cleanUpLevel (SymbolTab, 1);
         }
-   | addressmod
+   | addressmod {}
    ;
```

Note the extra curly braces after `addressmod`.

For vs2015 there have been a few places where variable declarations had to be hoisted to the top to be able compile.

And I've had problems with the compiler unable to find some files so likely a few files are missing from some solutions.  When I first was trying to compile I think I just copied those files to where they were needed, not really the right thing to do.

aralbrec, 2018-05-13:
I did an svn checkout of the current sdcc.  This is my diff file for vs2015:
https://drive.google.com/file/d/1ZNCnlX7edbgfxieH1nIWL677PiNwlurw/view?usp=sharing

The vcxproj files mainly upgrade the toolset from vs2010 to vs2015.  There are some dependencies in there on the location of boost in my machine.

The external dependencies are on bison and flex.  I don't remember where I got those for win32.

With these changes only two exes fail to compile for me due to headers that can't be found which should be easy enough to fix.

pauloscustodio, 2018-05-23:
Where is the final list of opcodes for the zxn, including the T-states of each instruction? In issue #312?

pauloscustodio, 2018-05-24:
sdas modified to accept ZXN opcodes; not yet done: copper unit and dma.

[sdas-zxn-patch.txt](https://github.com/z88dk/z88dk/files/2037122/sdas-zxn-patch.txt)

[f1.txt](https://github.com/z88dk/z88dk/files/2037130/f1.txt)

pauloscustodio, 2018-06-02:
@aralbrec, modified sdas to accept zxn opcodes and copper unit opcodes.
How do you manage the patches to sdcc? 

patch file: 
[sdas-zxn-patch.txt](https://github.com/z88dk/z88dk/files/2065426/sdas-zxn-patch.txt)

spth, 2018-06-03:
https://sourceforge.net/p/sdcc/patches/

Philipp

pauloscustodio, 2018-06-03:
Thanks, @spth. I've submitted the patch.

spth, 2019-01-03:
I had to change src/SDCC.y like so:
> ```
> ===================================================================
> --- src/SDCC.y	(revision 9958)
> +++ src/SDCC.y	(working copy)
> @@ -182,7 +182,7 @@
>            allocVariables ($1); [...]

What happened without the braces? Was there an error message from yacc / bison? I yes, which one?

Philipp

feilipu, 2019-02-11:
Obsolete? Resolved in sdcc.

closed 2019-02-11 by suborb