z88dk/z88dk #693: sccz80

aralbrec, 2018-04-14:


suborb, 2018-04-14:
Fastcall generation has always been available in sccz80. It does it in a weird way - pushes the parameter into the stack as the first thing in a function but at least that saves the overhead of pushing at the caller level. 

The char issue is probably just a missing case as a result of the decl rewrite in the autumn. I’ll get it fixed up this evening. 

I’ve not done much on the 32 bit code in the recent work - beyond all obvious optimisations back in the autumn  - most of it is still farmed out to the library routines. If you can provide the source and specific examples I can take a look.

aralbrec, 2018-04-14:
I'll email you a link to the source since he doesn't want it in public.  It's a zx next program and needs esxdos emulation so you'll have to use a next emulator.  [cspect](http://dailly.blogspot.ca/2018/02/cspect-110.html) is the easiest to get running - it's windows but it works under wine too.  zesarux is another choice and is a more complete emulation on the whole (sourceforge).

sccz80 compile
`zcc +zxn -subtype=sna -Cz"--main-fence=0xfdfd" -v -O3 -startup=24 -DTIMEX_HI_RES -clib=new --list -m src/emu.c src/main.c -o bin/sccz80/magnetic -create-app`

zsdcc compile
`zcc +zxn -subtype=sna -Cz"--main-fence=0xfdfd" -v -startup=24 -DTIMEX_HI_RES -clib=sdcc_iy -SO3  --list -m src/emu.c src/main.c -o bin/sdcc_iy/magnetic -create-app`

The sccz80 compile has problems getting started because of the fastcall but when it runs (prior to the fastcall change), it doesn't run properly.

You need two files in cspect's directory:  magnetic.sna and pawn.mag.

Run cspect with:

`cspect -zxnext -w3 -mmc=.\`

The `-w3` may not be necessary; this scales down the display so that it fits in my laptop screen.

Once started, press F2 to load the sna.

For simple testing try:

e
say hello
take note
ask about wristband
take chest
inv

aralbrec, 2018-04-18:
Fixed by #698

closed 2018-04-18 by aralbrec

suborb, 2019-03-26:
@aralbrec I've just reproduced this issue the other way round, with sdcc failing to print the description of the first room - have you had any similar reports?

aralbrec, 2019-03-29:
No - have you tried with the previous version 9958 ?  I haven't had time to look at the most recent but thre were numerous problems when I tried updating sdcc earlier.