z88dk/z88dk #695: newlib: crts should copy im2 tables into ram during initialization

aralbrec, 2018-04-17:
Coming from #677.

The newlib crts currently support four scenarios for im2 tables:

1. `__crt_org_vector_table = -0x80`  The im2 vector table is half size starting with vector 0x80 and is located at address 0x80 inside the library's page zero code.
2. `__crt_org_vector_table = -0x100`  The im2 vector table is full size and is located at address 0x100 inside the library's page zero code.
3. `__crt_org_vector_table = 0`  The im2 table is generated as a separate output binary that the user is responsible for placing in memory somehow (maybe appmake does it).
4. `__crt_org_vector_table = anything else`.  The im2 table is generated and stored in the output binary just after the crt.  But it is not copied to its destination address in ram - that is the user's responsibility.

crt4 is set up for no.4 above by default `defc TAR__crt_org_vector_table      = 0xFFF0`.  But I don't see anything that will copy the im2 table into ram.  There should be some code inserted into `code_crt_init` that copies the table to the destination address, something like:

```
ld hl,__interrupt_vectors_head
ld de,__crt_org_vector_table
ld bc,__interrupt_vectors_size
ldir
```

Maybe this is something all crts in z88dk should be doing by default when case 4 is chosen.

Edit:  I think case 4 should maybe be split.  If the org is negative, the crt will not copy the table to ram but if the org is positive, the crt will.

feilipu, 2018-04-17:
I found the [code that did the copying for me](https://github.com/z88dk/z88dk/blob/667cc8b24dbf52f07563f940b04b3198e7c3b3c9/libsrc/_DEVELOPMENT/target/crt_set_interrupt_mode.inc#L1), and the code works.

BTW. I'm not 100% certain what is the problem with defb __crt_org_vector_table&$F0?
Is it that __crt_org_vector_table is fixed during the library compile, and not the end user compile?

aralbrec, 2018-04-17:
Yeah I found it too after looking through your crt when you said it would work.  I was thinking the easiest way to do it would be to add to `crt_interrupt_mode.inc` and there it was :)

But I think adding the negative option will fill another need where someone wants the interrupt vectors available in the binary but not transferred to ram.

aralbrec, 2018-04-19:
The library already behaves as suggested above.  For case 4, if `__crt_org_vector_table < 0` then the interrupt table is stored in the output binary but it is not copied to ram at startup.  If `__crt_org_vector_table > 0` then the interrupt table is stored in the output binary and it is also copied to ram at startup.

commit c3399b7c (2018-04-19): Fix #695: newlib im2 vector table < 0 causes im2 table not to be copied to ram at startup
closed 2018-04-19 by aralbrec