z88dk/z88dk #697: [C128][T64][C128 mode] Computer freezes after tape load without starting the program

Fabrizio-Caruso, 2018-04-17:
I compile with
```
c128_z80_80col:
	$(Z88DK_PATH)$(MYZ88DK) +c128 \
	-D__CPM_VT100__ -D__C128_Z80__ \
	-DTINY_GAME -DLESS_TEXT -DNO_SLEEP -DNO_WAIT \
	$(SOURCE_PATH)/display_macros.c \
	$(SOURCE_PATH)/enemy.c \
	$(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c $(SOURCE_PATH)/text.c  \
	$(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c \
	$(SOURCE_PATH)/main.c \
	-create-app
```
I get an "A.T64" file which I try to load with Vice (either with autoload or by attaching the image and running the load command and starting the tape).
Loading starts, them I get a READY message, no cursor and a frozen C128.

zx70, 2018-04-17:
could you please try to build a smaller program ?   that behaviour looks the typical hang consequent to a stack or memory overflow

Fabrizio-Caruso, 2018-04-17:
@zx70 sure, I will and report here but I compiled my game with the TINY option. The T64 is barely 7k.

Fabrizio-Caruso, 2018-04-17:
C128 for mode 128 seems broken to me (if Vice is accurate enough).
I get the same problem with a simple hello world:
```
#include <stdio.h>

int main()
{
	printf("Hello world!\n");	
	while(1)
	{
		if(kbhit())
		{
			printf("%c\n",getch());
		}
	}			
	return 0;
}
```

zx70, 2018-04-18:
I just tried it,  and it seems in working order.
I compiled the plottest.c example, currently discussed (-lgfx128,  lo rez) and got A.T64.
I launched x128 (part of the vice emulator)
then I chose "aurotun dsk or tape".. it found program "A" and, after pushing ctrl (?)   I got the Ready message
I typed LOAD again, to pick the prebuilt sys command
RUN

works.

Fabrizio-Caruso, 2018-04-18:
I type LOAD
then I get till the READY message with no cursor.

Fabrizio-Caruso, 2018-04-18:
The ZIP below contains A.T64 with the hello world code. I am not able to load it.

[A.zip](https://github.com/z88dk/z88dk/files/1925883/A.zip)

zx70, 2018-04-19:
I'm using VICE 1.22, 32 bit.
[image]
[image]

Fabrizio-Caruso, 2018-04-19:
It does work on old Vice 1.22 but no longer on Vice 1.31. Who is the culprit?
Should I open an issue to the Vice developers?

zx70, 2018-04-19:
Fabrizio, I think the problem is on VICE but obviously we can't be 100% sure.
I picked another suggestion from the online forums, it can be that loading from tape images is handled by trapping the ROM functions.   If so the traps could fail with some KERNAL version, so  you can still try to compare the two emulator configurations (nationality, etc..) and check if they are exactly the same.

Fabrizio-Caruso, 2018-04-19:
Thanks @zx70 , I am personally a fan of TAPDancer, which is an Android app. I transform my Quad Core LG 6 smartphone into a humble 1530 cassette player.

Fabrizio-Caruso, 2018-04-19:
Did I miss:
"NOTE: a bug in the emulator will prevent this method to work with VICE version 3.x "
in the doc? Has it just been edited?

If it was already known, it means it is a known issue in Vice 3.x.

zx70, 2018-04-19:
I totally forgot that note, sorry for the lack of detail.
I'm closing the issue.

closed 2018-04-19 by zx70