z88dk/z88dk #699: [8080/8085 support][question] Integration of Small C - 8085 ?

Fabrizio-Caruso, 2018-04-18:
It seems that there are some (sort of) Small C (+ some ANSI features) 
cross compilers for 8080 and 8085 CPUs.

I have found 
https://github.com/ncb85/SmallC-85
which produces 8080/8085 assembly. 

How possible would it be to extend Z88DK to 8080/8085 CPUs?
8080 Assembly uses a different syntax, as far as I know. A convertion to Z80 syntax should not be hard, though.
I guess the major problem would be to get the lib to work with fewer registers...

suborb, 2018-04-18:
We gave up trying to support the 8080 years ago - it ends up being just too restrictive, the code wasn't removed though.

So, if you fancy giving it a go you'd need to:

- Avoid using longs and doubles
- Compile with -Cc--noaltreg flag -O0 --opt-code-speed=none
- Create versions of the routines z80_crt0s/crt0/ that avoid 8080 routines (e.g. shift, multiply etc)

You may need to adjust the compiler so it knows about CPU_8080 and avoids inlining shifts etc

Fabrizio-Caruso, 2018-04-18:
@suborb THANKS for the info!

closed 2018-04-18 by Fabrizio-Caruso

feilipu, 2018-04-18:
Just for interest, I converted one of the earliest 8080 libraries the [Lawrence Livermore Laboratories floating point library](https://github.com/feilipu/LLL-Floating-Point) to Z80 code using the tool in z88dk. The conversion was pretty much perfect. Just [two small errors](https://github.com/feilipu/LLL-Floating-Point/commit/b30d59f84afe3667187d0e9c5634b93ae3c00ed0). So 8080 code is supported by z88dk through this path.

Fabrizio-Caruso, 2018-04-18:
@feilipu You mean that 8080 can be used but that should always be the case since it is a subset of Z80. Isn't it?
What I ask is whether I can build 8080 or 8085 binaries. The goal is to support targets with 8080 and 8085 CPUs that do not have Z80 extensions.
I understand that the libs would not work but if an 8080/8085 binary can be compiled and linked, it would be a great thing.

feilipu, 2018-04-18:
There is full binary compatibility. Here's a [mapping table](http://nemesis.lonestar.org/computers/tandy/software/apps/m4/qd/opcodes.html), for reference.

It shows the 8080 and the Z80 equivalent instruction. The opcode is the same for both CPUs.

By extension, if you start with 8080 instructions, and then migrate these to Z80 instructions, the assembler will produce Z80 opcodes which are 8080 compatible opcodes.

zx70, 2018-05-04:
FYI, also the z80 to 486 conversion is fairly accurate.  I was able to port a simplified integer BASIC interpreter for the 8085 to MSDOS (486 CPU or later only), the resulting binary was very small (3..4 Kbytes including some extra stuff, like a BIOS call interface for graphics and SAVE / LOAD support).
I've never seen a BASIC interpreter running that fast !