z88dk/z88dk #702: (z80asm) UNDEFINE of a PUBLIC constant doesn't work

suborb, 2018-04-21:
The following code won't assemble:

                PUBLIC  CONSOLE_COLUMNS
                defc    CONSOLE_COLUMNS = 80

                UNDEFINE CONSOLE_COLUMNS
                defc    CONSOLE_COLUMNS = 32

                defb    CONSOLE_COLUMNS

Reporting the error:

Error at file 'def.asm' line 7: symbol 'CONSOLE_COLUMNS' already defined

Move the PUBLIC statement to after the UNDEFINE resolves the problem.

suborb, 2018-04-22:
I've worked around this by jiggling the order of the exports.

feilipu, 2026-11-09:
UNDEFINE clears a PUBLIC constant and keeps the declaration. A later defc replaces the value. The symbol stays public. Commit 7a0f28ccab.

closed 2026-11-09 by feilipu