z88dk/z88dk #715: sinwave/coswave examples

zx70, 2018-04-27:
those 2examples had simple acceleration based on the picture symmetry .. not working anymore.

suborb, 2018-04-27:
Odd:

[image]

suborb, 2018-04-27:
First few coordinates with math48:

(0,214)
(-255,214)
(0,213)
(-255,213)
(0,179)
(-255,179)

with genmath:

(-1,214)
(-256,214)
(-1,213)
(-256,213)
(-1,179)
(-256,179)

On my mac:

(0,214)
(255,214)
(0,213)
(255,213)
(0,179)
(255,179)

I think this is in my court so I'll look at it.

suborb, 2018-04-27:
Fixed - one of the branches wasn't swapping over the left and right hand sides of the subtraction when a constant was involved.

It's times like this that you realise how slow floating point on a z80 is:

```
% gcc sinwave.c
% time ./a.out > nix.out
0.004u 0.001s 0:00.00 0.0%	0+0k 0+0io 0pf+0w

% zcc +test sinwave.c -lm
% time ticks -w 600 a.bin > test.out
20.812u 1.281s 0:22.56 97.9%	0+0k 0+0io 56pf+0w

% zcc +test sinwave.c -lmath48
% time ticks -w 600 a.bin > test.out
18.615u 1.239s 0:20.22 98.1%	0+0k 0+0io 0pf+0w

% zcc +test sinwave.c -lmath48 -compiler=sdcc
% time ticks -w 600 a.bin > test.out
17.422u 1.098s 0:18.84 98.2%	0+0k 0+0io 2pf+0w
```

[image]

closed 2018-04-27 by suborb