z88dk/z88dk #716: z80asm: new directive DISP

aralbrec, 2018-04-27:
We have a PHASE directive which allows the absolute address of code to be changed temporarily. This allows code to be assembled to a different address than where it is stored.

We're missing a similar directive DISP that alters the assemble address by a relative amount rather than an absolute amount.

The need is coming from the zx next target where physical memory containing code/data is separate from logical address.  Suppose you have an 8k page PAGE_N.  This page can be mapped to any of the 8k slots of the z80's address space, ie in 0-0x1fff, 0x2000-0x3fff, 0x4000-0x5fff, ... , 0xe000-0xffff.  Suppose you want to fit two different subroutines in that 8k page, one (function A) run when in slot 0x2000-0x3fff and the other (function B) when in slot 0xe000-0xffff.  The problem is to pack both subroutines into PAGE_N with different org addresses within the 8k slot.

Right now you can place the two subroutines with absolute org into PAGE_N.  But the org must be absolute so it's a manual process for packing the two routines into PAGE_N.  To do this the user creates some sections to hold the routines inside PAGE_N:

```
SECTION PAGE_N_FUNC_A
org 0x2000   ;; inside mmu1 0x2000-0x3fff
SECTION PAGE_N_FUNC_B
org 0xe200   ;; inside mmu7 0xe000-0xffff
```

and then this is included into the crt's memory map with a pragma.  Appmake understands the zx next's memory and will place sections with names containing "PAGE_N" into physical page N.  To do this, because physical pages are 8k in size, it logically ANDs the org of any sections within the page with 0x1fff to generate the offset where the section will be stored physically in the page.
 
In the above example, `SECTION PAGE_N_FUNC_A` will be stored at offset `0x2000 & 0x1fff = 0` within PAGE_N and `SECTION PAGE_N_FUNC_B` will be stored at offset `0xe200 & 0x1fff = 0x200` within PAGE_N.  If PAGE_N is placed in mmu1 (the 0x2000-0x3fff region) then calling function A at address 0x2000 will work fine.  Likewise, if PAGE_N is placed in mmu7 (the 0xe000-0xffff region) then calling function B at address 0xe200 will also work fine.

The great weakness here is the manual org that is required to pack code with different org into a physical memory page.  You have to make sure things don't overlap and you don't want to waste memory by having gaps appear between code chunks.  It's manual linking.

To get z80asm to pack code with different org into a physical page, the proposal is for a **DISP** directive.

The above example with DISP would look like:

```
SECTION PAGE_N   ;; supplied by the crt
org 0xc000   ;; supplied by the crt or user

SECTION PAGE_N_FUNC_A
disp 0x6000  ;; effective assemble address = current_asm_address + 0x6000 = offset_from_start_of_PAGE_N  at 0xc000 + 0x6000 = offset from 0x2000 in mmu1

SECTION PAGE_N_FUNC_B
disp 0x2000  ;; effective assemble address = current_asm_address + 0x2000 = offset_from_start_of_PAGE_N  at 0xc000 + 0x2000 = offset from 0xe000 in mmu1
```

Some rules:  to disable disp, use `disp 0`.  If an org is met, set disp=0.  Alignment must be applied to current asm address + disp.

Doing this, the linker will automatically pack functions A and B into physical PAGE_N and both functions will have org set correctly to run out of mmu1 and mmu7 respectively.

pauloscustodio, 2018-05-22:
@aralbrec, do you have some pointers to docs describing the zxn mmu? I miss some background information.

aralbrec, 2018-05-23:
I'd like to think some more on disp before moving forward.

There are a few different memory spaces.  A lesser used one will be divmmc memory which can only be paged into the 8k-16k area (except for one page which can appear in 0-8k).  The ROMs are in another memory space too but obey the standard ports 1ffd,7ffd to determine what appears in the bottom 16k.

The main memory space we're talking about is the rest of ram called the BANK space (if divided into 16k chunks compatible with the zx128's view of memory) or the PAGE space (if divided into 8k chunks and is the zx next's view of memory).  There are up to 224 8k PAGEs numbered 0-223.

The z80's 64k address space is divided into 8k slots called mmus.  So mmu0 corresponds to the 0-8k region, mmu1 to 8k-16k, ... mmu7 for 56k-64k.  There is an instruction to place any 8k page into an mmu slot.  So "mmu1 40" would place 8k page 40 into mmu1 which is the 8k-16k region.  You can make page 40 appear in more than one mmu slot "mmu7 40" would also make 8k page 40 appear in the 56k-64 region.

As a result there is a distinction between physical storage of a function (what page is it in?) and its org (where does it appear in the z80's address space?).  Appmake has already been updated to handle this situation very well.  The model has the user place code or data into physical storage using sections defined in the memory map named, eg, "PAGE_0" through "PAGE_223".  Then for placement of the page in the z80's 64k address space, the org for pages can be changed via pragma.  So "#pragma output CRT_ORG_PAGE_40 = 0x2000" would indicate that page 40 will be placed in mmu1 by the program when accessed.  Because the hardware also supports the 128k's view of memory as divided into 16k banks, the memory map also defines "BANK_0" through "BANK_111" which corresponds to the same memory as pages but are 16k in size.  Appmake merges stuff placed into BANKs and PAGEs together and indicates if there are overlaps.  It also enforces 16k size limits on BANKs (and 8k size limits on BANK_L, BANK_H) to be consistent with port 7ffd banking but there are no such restrictions on PAGEs where overflow will simply mean stuff spills into following pages.

Then appmake makes the section substrings "BANK_N" and "PAGE_N" special in that if the user defines his own sections with those substrings, appmake will place that sections inside physical "BANK_N" or "PAGE_N".

So the user can add to the memory map:

SECTION PAGE_90_MUSIC
org 0x5000
SECTION PAGE_90_FX
org 0xc000

and appmake will place both those sections into physical PAGE_90.  The offsets into PAGE_90 are the ORG addresses logically & with 0x1fff.

It's this last capability that's missing - being able to put multiple org into one page without doing it manually.  In the above example, PAGE_90_MUSIC is expecting to be seen in mmu2 (0x4000-0x5fff) with an offset of 0x1000 into PAGE_90.  PAGE_90_FX is expecting to be seen in mmu6 (0xc000-0xdfff) with an offset of 0x0000 into PAGE_90.  The manual org are placing these two sections within physical page 90 and there is no way to have this done automatically so that there are no gaps or overlaps.

This is a less common use-case but it may be something interesting to add.